react-native-camera/react-native-camera #2089: Prefix attr value to prevent conflicts

jbcullis, 2019-02-03:
# Feature-Request

**Describe the Feature**
I recently ran into the issue of a conflict with the attr value aspectRation where the square sdk also had an entry however with a different format.

**Possible Implementations**
It would be nice to prefix any attr values to clearly identify they belong to react-native-camera and prevent conflicts with other packages.

**Related Issues**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this feature to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

sibelius, 2019-02-03:
Can you send me the link of this sdk?

Does this sdk uses react native camera?

jbcullis, 2019-02-03:
This is the sdk - https://github.com/square/react-native-square-reader-sdk

I don't think it uses the camera for taking photos, but it does do image uploads for logo's and product photos.
It may also reference the aspect ratio for the signature pad?

Either way, I though it would be much easier for libraries to prefix their variables to prevent duplicate entries.

sibelius, 2019-02-03:
What if another library conflicts with the prefixed version?

This can’t scale, we won’t change because of this reason

closed 2019-02-03 by sibelius

sibelius, 2019-02-03:
aspectRatio={props.sdkAspectRatio}