react-native-camera/react-native-camera #2093: Support for camera shutter speed?

Xeoncross, 2019-02-04:
# Feature-Request

**Describe the Feature**

Both Android and iPhone lenses support specifying shutter speed. By default they calculate the speed for you along with ISO and Aperture. It would be great to have access to camera shutter speed as an option.

On Android this seems to be be handled by reading and writing `Camera.Parameters`. I'm not sure how iOS handles this.

**Possible Implementations**

Unfortunately, it looks like different phones [handle this differently](https://stackoverflow.com/a/37567225/99923) so you have to parse the camera parameters to see what options should be shown on the UI. (Example Android Code)

            Camera mCamera = Camera.open();
            Camera.Parameters params = mCamera.getParameters();

            String flatten = params.flatten();
            String[] paramsSeparated = flatten.split(";");
            for(String cur : paramsSeparated) {
                if(cur.contains("iso") && cur.contains("values")) {
                    ...
                }
            }

closed 2019-06-08 by stale[bot]

sejas, 2019-06-14:
I think it's a great idea.
In my case, I need to increase the shutter speed to get less burry images.

cinder92, 2020-03-07:
any update on this issue?