react-native-camera/react-native-camera #2097: Image size of IOS is cropped even if the view is in full screen

moribaleta, 2019-02-08:
# Question

**To Do First**
- [ ] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [ ] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [ ] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**

**Tags**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

moribaleta, 2019-02-08:
the image resolution in Android is the same with the Android Camera App
but in IOS if the image is taken the output image resolution has lower height than the native IOS camera app

//'use strict';
import React from 'react';
import { Dimensions, StyleSheet, TouchableOpacity, TouchableHighlight, View, Image, ActivityIndicator } from 'react-native';
import SnackBar from 'rn-snackbar'
import { RNCamera } from 'react-native-camera';
import { HeaderButtonIconTemplate } from '../components/button_template'
import { ADD_IMAGE, GO_BACK, VIEW_GALLERY } from '../../store/image_store/image_action';
import { connect } from 'react-redux';
import { screen_size } from '../components/screen_size';

class PatientProfileCamera extends React.Component {
    constructor(props) {
        super(props);
        this.state = {
            path: null,
            images: null,
            isTaking: false
        };
        this.view_image = false
    }

    onLayout = (e) => {
        const { width, height } = Dimensions.get('window')
        console.log(width, height)
        this.forceUpdate()
    }

    static navigationOptions = ({
        navigation
    }) => {
        return {
            header: () => null
        }
    }

    componentWillReceiveProps(props) {
        this.setState({ path: props.curr_image })
    }

    _takePicture = () => {
        if (!this.state.isTaking) {
            try {
                if (this.camera) {
                    this.setState({ isTaking: true })
                    this.camera.takePictureAsync().then(data => {
                        console.log("image taken %o", data)
                        this.props.add_image(data.uri)
                        this.setState({ isTaking: false })
                        Image.getSize(data.uri, (width, height) => {
                            console.log("width: %o -- height: %o", width, height)
                        })
                    }).catch(err => {
                        this.setState({ isTaking: false })
                        throw err
                    })
                } else {
                    throw 'camera not available'
                }
            } catch (err) {
                console.log('err: ', err);
                SnackBar.show('error ' + err, {
                    confirmText: 'Close',
                    onConfirm: () => {
                        SnackBar.dismiss()
                    },
                    style: { opacity: 0.95 },
                    backgroundColor: '#33A0AF',
[... 137 lines cut ...]
        width: '100%',
    }
});

const mapStateToProps = (state, ownProps) => {
    return {
        curr_image: state.images[state.images.length - 1]
    }
}

const mapDispatchToProps = dispatch => {
    return {
        add_image: (image) => {
            dispatch(ADD_IMAGE(image))
        },
        view_gallery: () => {
            dispatch(VIEW_GALLERY)
        },
        goBack: () => {
            dispatch(GO_BACK)
        }
    }
}

export default connect(mapStateToProps, mapDispatchToProps)(PatientProfileCamera)

moribaleta, 2019-02-08:
[image]

the right side is the output from the Native IOS Camera app on the right was the result from the RNCamera

moribaleta, 2019-02-08:
[image]

closed 2019-06-08 by stale[bot]

GaspardC, 2021-02-10:
@moribaleta I have the same issue, did you find a solution ?

moribaleta, 2021-02-10:
> @moribaleta I have the same issue, did you find a solution ?

i made my own

GaspardC, 2021-02-15:
@moribaleta awesome,  could you share with me how you did it?

moribaleta, 2021-02-16:
> @moribaleta awesome, could you share with me how you did it?
@GaspardC 
https://github.com/moribaleta/ReactNative-NativeCameraIOS-Template