react-native-camera/react-native-camera #2100: Face detection yaw angle always 0 and front facing camera not detecting face

ManigandanRaamanathan, 2019-02-11:
# Bug Report
**To Do First**
- [Yes] Did you try latest release?
- [No] Did you try master?
- [No] Did you look for existing matching issues?

**Related Modules**
RNCamera
FaceDetector

**Platforms**
Android

**Description/Current Behaviour**
Face detection yaw angle always 0 on back camera and front facing camera is not detecting face

**Expected Behaviour**
Should get yaw angle and front facing camera should detect face

**Steps to Reproduce**
Used example code provided for face detection

ManigandanRaamanathan, 2019-02-12:
@n1ru4l guys any update on this issue? Can you please fix these 2 bugs on yaw angle value and front camera face detection?

n1ru4l, 2019-02-12:
Please feel free to send a pull request to fix those issues. I have neither the time or the requirements to fix this behaviour. If you or your company rely on react-native-camera you might consider allocating time for working/improving on this library.

prodoxx, 2019-02-15:
Hey @ManigandanRaamanathan , try installing React-Native Camera version 1.2.0 (or a lower version that is not latest). I got the latest version a yesterday and the face detection was not working properly. It was not detecting faces properly or it looked like it was too slow. I downgraded to 1.2.0 and it works fine now.

ManigandanRaamanathan, 2019-02-15:
@prodoxx thanks for the info. Will try the older version and update here

ManigandanRaamanathan, 2019-02-15:
@prodoxx is the front / selfie camera detects face on the version that you're using?

prodoxx, 2019-02-15:
Yeah, I'm using version 1.2.0 right now and it detects it.

ManigandanRaamanathan, 2019-02-15:
@prodoxx thanks for the update.

darknblack, 2019-02-22:
Not working to me as well. It used to work but after updating react native from version .57.4 to 58.5 it stopped working. tried downgrading but still no good.

darknblack, 2019-02-22:
Downgraded to 1.3.1 solves this problem.

sibelius, 2019-02-22:
anybody can do a git bisect and check when there was this regression?

darknblack, 2019-02-22:
> anybody can do a git bisect and check when there was this regression?

I checked my package-json.lock before and after I update the packages and here's what i found out.

1.12.0 = facial recognition working but back camera only
1.3.1 = facial recognition working both back and front camera

williamgurzoni, 2019-03-14:
Same problem. I downgrade to 1.3.1 and it works, but yawAngle don't.

sibelius, 2019-03-14:
@williamgurzoni can you try v2 using MLKit

HorusGoul, 2019-03-22:
@sibelius 

v2 seems to have this issue as the face detector doesn't detect my face properly with the front camera

Avdhutk, 2019-04-15:
> @sibelius
> 
> v2 seems to have this issue as the face detector doesn't detect my face properly with the front camera

Any updates on this? Facing same issue with V2.3.0.

sibelius, 2019-04-15:
@Avdhutk can you debug on native code?

Avdhutk, 2019-04-15:
> @Avdhutk can you debug on native code?

Yes

Avdhutk, 2019-04-15:
I need to use MLKit but facing problem in face detecting by front camera with latest version. 
Front camera not returning correct bounds and eye positions. Stucked due to this.

o-alexandrov, 2019-04-17:
Please forgive the comment that might seem as a 👍 alternative.
Just wanted to ask the maintainers for the preferable reproduction example for you to see that the front facing camera is not capable of doing face detection.

Personally, I didn't try the MLKit, but precisely followed every single step for the default face detection.

closed 2019-05-09 by sibelius

n1ru4l, 2019-05-09:
:tada: This issue has been resolved in version 2.7.0 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera)
- [GitHub release](https://github.com/react-native-community/react-native-camera/releases/tag/v2.7.0)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

PR #2257 (2019-05-09): Fix android front camera face detection
## Intro

Android front camera face detection has been broken since this PR #2050 had landed I guess and released within **1.8.1** version.

Fixes #2100, #1962, #2246, #2221 maybe #1325

## Tested devices: 

- Pixel 2 (Android Q)
- Samsung S7 (Android 8.0)
- Nexus 5 (Android 6.0.1)