react-native-camera/react-native-camera #2105: Auto focus not working with camera2 api.

jaresan, 2019-02-13:
# Bug Report
**To Do First**
- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Related Modules**
RNCamera

**Platforms**
Android

**Versions**

- Android: 8.0
- react-native-camera: 10.1.2 & master
- react-native: 0.58.4
- react: 16.6.3

**Description/Current Behaviour**
When using `useCamera2Api` prop on Camera, the `AutoFocus` mode doesn't work.
I've read quite a few issues regarding the slow image capture on Android and noticed with camera2Api it works much faster. Focus stopped working though.

**Expected Behaviour**
Faster picture taking by using camera2Api without omitting focus.

**Steps to Reproduce**
See code below. Install `react-native-camera` and use camera2Api.

**Does it work with Expo Camera?**
When using camera2Api with Expo Camera, the pictures can be taken faster and in focus, works without a problem.

**Additionals**

```
import React, {Component} from 'react';
import { StyleSheet, Text, View, TouchableOpacity, Dimensions, Image } from 'react-native';
import { RNCamera } from 'react-native-camera';

export default class App extends Component {
  state = {};

  renderApp = () => [
    <RNCamera
      ref={ref => {
        this.camera = ref;
      }}
      useCamera2Api
      autoFocus={RNCamera.Constants.AutoFocus.on}
      style={styles.camera}
      type={RNCamera.Constants.Type.back}
      flashMode={RNCamera.Constants.FlashMode.off}
      permissionDialogTitle={'Permission to use camera'}
      permissionDialogMessage={'We need your permission to use your camera phone'}
      onGoogleVisionBarcodesDetected={({ barcodes }) => {
        console.log(barcodes);
      }}
    />,
    <View style={{ flex: 0, flexDirection: 'row', justifyContent: 'center' }}>
      <TouchableOpacity onPress={this.takePicture} style={styles.capture}>
        <Text style={{ fontSize: 14 }}> START </Text>
      </TouchableOpacity>
    </View>
  ];

  renderImage = () => (
    <View>
      <Image
        source={{ uri: this.state.imageUri }}
        style={styles.preview}
      />
    </View>
  );

  render() {
    return (
      <View style={styles.container}>
        {this.state.imageUri ? this.renderImage() : this.renderApp()}
      </View>
    )
  }

  takePicture = async () => {
    if (this.camera) {
      const options = { quality: 1, skipProcessing: true };
      const { uri } = await this.camera.takePictureAsync(options);
      this.setState({
        imageUri: uri
      })
    }
  };
}

const styles =...;```

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

SimonErm, 2019-02-18:
Workaround: disable autofocus, then it works^^. No kidding 😆

sibelius, 2019-02-18:
can you guys test this PR https://github.com/react-native-community/react-native-camera/pull/1974

maybe it solves your problem

closed 2019-06-08 by stale[bot]