react-native-camera/react-native-camera #2110: Deprecate `permissionDialogTitle` and `permissionDialogMessage` in favor of `androidCameraPermissionOptions` and `androidRecordAudioPermissionOptions`

n1ru4l, 2019-02-15:
# Feature-Request

**Describe the Feature**
Instead of having different properties like:

- `permissionDialogTitle`
- `permissionDialogMessage`
- `permissionButtonPositive` (not implemented yet)
- `permissionButtonNegative` (not implemented yet)
- `permissionsButtonNeutral` (not implemented yet)

We should have a better approach:

- `androidCameraPermissionOptions`
- `androidRecordAudioPermissionOptions`

This would solve the following issues:

- Show different text/dialogues for requesting camera permissions and requesting audio record permissions
- We reduce the amount of needed properties and do not need to add new properties in case the properties of [PermissionsAndroid](https://facebook.github.io/react-native/docs/permissionsandroid) change

**Possible Implementations**
- Deprecate `permissionDialogTitle` and `permissionDialogMessage` aka show warning
- Ignore `permissionDialogTitle` and `permissionDialogMessage` if either `androidCameraPermissionOptions` or `androidRecordAudioPermissionOptions` is set

**Related Issues**

#1849

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this feature to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

kulak-at, 2019-04-12:
Hey, I'm working on PR for this issue.

kulak-at, 2019-04-15:
This task is now resolved in 2.4.0 after #2206 has been merged.

closed 2019-04-23 by n1ru4l