react-native-camera/react-native-camera #2116: Allow user to choose flash mode while the camera is open

jonahgreenthal, 2019-02-18:
# Feature-Request

**Describe the Feature**
When the camera screen is active, the user should be able to choose the flash mode. (If the developer wants to let them.)

**Possible Implementations**
iOS displays a "lightning bolt" icon at the top left of the native camera screen, which when tapped expands a menu (Auto, On, or Off). Imitating that implementation seems reasonable.

We could add an additional `RNCamera.Constants.FlashMode.userChoose` constant to represent this possibility.

sibelius, 2019-02-19:
Can’t this be implemented on user land?

jonahgreenthal, 2019-02-19:
What does "user land" mean? Do you mean that the developer using `react-native-camera` should implement this themself? (By creating their own overlay on the `<RNCamera>` component?)

If so: I take your word that it is possible, but it still seems like it would be a convenient feature for the package to provide.

sibelius, 2019-02-19:
Packages should provide the building blocks to users implements their features 

We should focus on fewer public apis to be easier to maintain

We can accept this as a recipe for the docs

SimonErm, 2019-02-19:
Take a look here: https://github.com/react-native-community/react-native-camera/blob/master/examples/basic/App.js

n1ru4l, 2019-02-20:
This should be handled in Userland as @sibelius said. Not everyone wants the same UI for their Camera Screen. If you need help for getting started check out the example app (which @SimonErm referred to).

I am going to close this issue. In case you still have questions related to react-native-camera I encourage you to open a new issue or question on stackoverflow.

closed 2019-02-20 by n1ru4l