react-native-camera/react-native-camera #2118: Would you like to add an option that set the maximum size of the result picture?

zhuhengtan, 2019-02-20:
Mostly we take pictures and upload, but with android phones' cameras upgrading, the size of picture is  rising too.It takes a very long time to upload the picture even set the quality to 0.1. I was an iOS developer so I know there are functions to shrink the size of picture, but I am not familiar with android. SO I am wondering if it's possible you add this option, or do you have any suggestions?
Thanks.

n1ru4l, 2019-02-20:
Do you mean something like this: https://stackoverflow.com/a/12602329/4202031

For Android?

zhuhengtan, 2019-02-20:
> Do you mean something like this: https://stackoverflow.com/a/12602329/4202031
> 
> For Android?

Exactly!

n1ru4l, 2019-02-20:
IMHO we should solve #1836 first before introducing any options that make the capture image process even slower as it is right now.

zhuhengtan, 2019-02-21:
Alright just follow your own steps.I am trying to solve this requirement by javascript on base64, though its a little slow😅