react-native-camera/react-native-camera #2122: [RCTCamera] scaling down images

pavermakov, 2019-02-21:
# Question

**To Do First**
- [+] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [+] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [+] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**
Due to the legacy state of the project I am working on, I am forced to use the old RCTCamera module. On iphoneX emulator images come out really big (2160 by 3840). Is there a way to scale images down?

Kielan, 2019-02-21:
This is also an issue with RNCamera module and I think should be discussed openly on a main issue thread :)

SimonErm, 2019-02-22:
I think on RCTCamera you can use the captureQuality Prop:
 https://github.com/react-native-community/react-native-camera/blob/master/docs/RCTCamera.md#capturequality 
On RNCamera you can do the same with the width option:
 https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md#takepictureasyncoptions-promise

closed 2019-04-18 by sibelius