react-native-camera/react-native-camera #2123: Face Detection not working on Android

witalobenicio, 2019-02-21:
# Bug Report
**To Do First**
- [ Yes] Did you try latest release?
- [ Yes] Did you try master?
- [ Yes] Did you look for existing matching issues?

**Related Modules**

RNCamera
FaceDetector

**Platforms**

Android

**Versions**

- Android: 27
- iOS:
- react-native-camera: 1.12.0 (tried with 1.2.0 too)
- react-native:  0.54.8
- react: 16.3.1

**Description/Current Behaviour**
TypeError: 'caller', 'callee', and 'arguments' properties may not be accessed on strict mode functions or the arguments objects for calls to them at Function.invokeGetter

**Expected Behaviour**
Detect faces

**Steps to Reproduce**

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

sibelius, 2019-02-21:
can you show the full stack trace?

witalobenicio, 2019-02-21:
@sibelius 
[image]
[image]

witalobenicio, 2019-02-21:
`02-21 11:13:54.913 616-616/? E/ANDR-PERF-MPCTL: invalid request, no optimizations performed
02-21 11:13:54.914 1232-1232/? E/QCamera: <HAL><ERROR> lock_acq: 366: failed to acquire lock
02-21 11:13:54.914 616-616/? E/ANDR-PERF-MPCTL: invalid request, no optimizations performed
02-21 11:13:54.914 1232-3591/? E/QCamera: <HAL><ERROR> lock_acq: 366: failed to acquire lock
02-21 11:13:54.924 963-3604/? E/mm-camera: <ISP   ><ERROR> 544: isp_port_process_upstream_event: failed: isp_handler_enqueue_event_to_trigger_queue
02-21 11:13:54.924 963-3604/? E/mm-camera: <ISP   ><ERROR> 599: isp_port_event: failed: isp_port_process_upstream_event
02-21 11:13:54.924 963-3604/? E/mm-camera: <PPROC ><ERROR> 678: base_port_handle_buf_divert_ack: base_port_handle_buf_divert_ack:678, failed
02-21 11:13:54.924 963-3604/? E/mm-camera: <IMGLIB><ERROR> 565: module_imgbase_forward_port_event: module_imgbase_forward_port_event_paaf:565] Fowarding event 2 from src port failed
02-21 11:13:54.924 963-3604/? E/mm-camera: <CPP   ><ERROR> 911: cpp_module_send_event_upstream: send_event_to_peer failed 0
02-21 11:13:54.924 963-3604/? E/mm-camera: <CPP   ><ERROR> 987: cpp_module_send_buf_divert_ack: [CPP_BUF:] send_event_upstream failed -14
02-21 11:13:54.927 963-3621/? E/mm-camera: <STATS_AIS ><ERROR> 173: dsps_send_req: DSPS Send Request Timeout!!
02-21 11:13:54.992 616-616/? E/ANDR-PERF-MPCTL: invalid request, no optimizations performed
02-21 11:13:54.992 1232-1232/? E/QCamera: <HAL><ERROR> lock_acq: 366: failed to acquire lock
02-21 11:13:55.001 877-24681/? E/CameraBgProcService: setHalCallback: ERROR: Callback not set
02-21 11:13:55.001 877-24681/? E/CameraBgProcService: setVfwCallback: ERROR: Callback not set
02-21 11:13:55.001 877-24681/? E/CameraBgProcService: setHalTouchCallback: ERROR: Callback not set
02-21 11:13:55.002 616-616/? E/ANDR-PERF-MPCTL: invalid request, no optimizations performed
02-21 11:13:55.003 1232-1232/? E/QCamera: <HAL><ERROR> lock_acq: 366: failed to acquire lock
02-21 11:13:55.010 963-3621/? E/mm-camera: <MCT   ><ERROR> 1448: mct_pipeline_decide_hw_wakeup: Couldn't find meta stream
02-21 11:13:55.042 963-3717/? E/quadracfa_dummy: inside dummy remosaic_deinit 1
02-21 11:14:30.295 616-616/? E/ANDR-PERF-MPCTL: invalid request, no optimizations performed
02-21 11:14:30.295 1232-1593/? E/QCamera: <HAL><ERROR> lock_acq_timed: 317: failed to acquire lock
02-21 11:14:30.327 963-3806/? E/mm-camera: <STATS_AIS ><ERROR> 173: dsps_send_req: DSPS Send Request Timeout!!
02-21 11:14:30.332 963-3819/? E/mm-camera: <IMGLIB><ERROR> 1180: module_drsd_start_session: module_drsd_start_session:1180] E
02-21 11:14:30.407 963-963/? E/mm-camera: <STATS ><ERROR> 3270: stats_port_check_caps_reserve: Invalid Port capability type!
02-21 11:14:30.407 963-963/? E/mm-camera: <STATS ><ERROR> 3270: stats_port_check_caps_reserve: Invalid Port capability type!
02-21 11:14:30.418 963-3827/? E/mm-camera: <MCT   ><ERROR> 1448: mct_pipeline_decide_hw_wakeup: Couldn't find meta stream
02-21 11:14:30.419 1232-3799/? E/MotCameraParameters: setReprocCount: dual-pass zoom disabled
02-21 11:14:30.419 877-877/? E/CameraBgProcService: setHalCallback: ERROR: Callback not set
02-21 11:14:30.419 963-3827/? E/mm-camera: <MCT   ><ERROR> 1448: mct_pipeline_decide_hw_wakeup: Couldn't find meta stream
02-21 11:14:30.419 877-877/? E/CameraBgProcService: setVfwCallback: ERROR: Callback not set
02-21 11:14:30.419 877-877/? E/CameraBgProcService: setHalTouchCallback: ERROR: Callback not set
02-21 11:14:30.423 1232-3833/? E/Thermal-Lib: Thermal-Lib-Client: No Callback registered for camcorder
02-21 11:14:30.423 963-3827/? E/mm-camera: <MCT   ><ERROR> 1448: mct_pipeline_decide_hw_wakeup: Couldn't find meta stream
02-21 11:14:30.423 963-3827/? E/mm-camera: <MCT   ><ERROR> 1448: mct_pipeline_decide_hw_wakeup: Couldn't find meta stream
02-21 11:14:30.450 963-3827/? E/mm-camera: <MCT   ><ERROR> 1448: mct_pipeline_decide_hw_wakeup: Couldn't find meta stream
02-21 11:14:30.455 616-616/? E/ANDR-PERF-MPCTL: invalid request, no optimizations performed
02-21 11:14:30.455 1232-1593/? E/QCamera: <HAL><ERROR> lock_acq: 366: failed to acquire lock
02-21 11:14:30.459 963-3827/? E/mm-camera: <IFACE ><ERROR> 2791: iface_util_modify_plane_info_for_native_buf: iface_util_modify_plane_info_for_native_buf: Invalid cam_format 2088533116
02-21 11:14:30.459 963-3827/? E/mm-camera: <IFACE ><ERROR> 2826: iface_util_calculate_frame_length_for_native_buf: iface_util_calculate_frame_length_for_native_buf: returned fail=-1
02-21 11:14:30.459 963-3827/? E/mm-camera: <IFACE ><ERROR> 2791: iface_util_modify_plane_info_for_native_buf: iface_util_modify_plane_info_for_native_buf: Invalid cam_format 2088533116
02-21 11:14:30.459 963-3827/? E/mm-camera: <IFACE ><ERROR> 2826: iface_util_calculate_frame_length_for_native_buf: iface_util_calculate_frame_length_for_native_buf: returned fail=-1
02-21 11:14:30.459 963-3827/? E/mm-camera: <MCT   ><ERROR> 1448: mct_pipeline_decide_hw_wakeup: Couldn't find meta stream
02-21 11:14:30.460 963-3827/? E/mm-camera: <MCT   ><ERROR> 1448: mct_pipeline_decide_hw_wakeup: Couldn't find meta stream
02-21 11:14:30.468 963-3827/? E/mm-camera: <IMGLIB><ERROR> 907: module_drsd_port_check_caps_reserve: module_drsd_port_check_caps_reserve:907] port already reserved
02-21 11:14:30.468 963-3827/? E/mm-camera: <IMGLIB><ERROR> 4550: faceproc_hw_comp_create: Library not loaded, handle=0x0, p_ops=0xab103c4c
[... cut ...]
02-21 11:14:30.676 963-3823/? E/mm-camera: <ISP   ><ERROR> 599: isp_port_event: failed: isp_port_process_upstream_event
02-21 11:14:30.812 963-3814/? E/mm-camera: <ERROR> 831: dmlroc_init: dmlroc_init cfg: camif 1944x2592, elem 80x80, thres 0.001600, accur 8
02-21 11:14:30.812 963-3814/? E/mm-camera: <ERROR> 832: dmlroc_init: dmlroc_init cfg: offsetH 16, offsetV 12
02-21 11:14:30.812 963-3814/? E/mm-camera: <ERROR> 833: dmlroc_init: dmlroc_init mesh: rxc 10x13, subs 4, hxw 54x54, offset vert 0, horz 0`

witalobenicio, 2019-02-21:
@sibelius any thoughts?

witalobenicio, 2019-02-21:
I think it may be a problem with my phone. I test on emulator and I don't get this error.

witalobenicio, 2019-02-21:
I just bought another phone and tested it. It is working.
But I think in Android 8.1 isn't.

darknblack, 2019-02-22:
Facing the same issues after migrating from .57.4 to .58.5 react native.

closed 2019-06-08 by stale[bot]