react-native-camera/react-native-camera #2127: barcode can not be recognized

ChenChuanzhen, 2019-02-28:
In my project, some barcode are not be recognized, such as
[image].How I can improve it?

ChenChuanzhen, 2019-02-28:
Does someone can tell me how I can improve it?

kunalkp, 2019-03-26:
We need negative case prevention like this. Even in my project, I am looking for something to handle when the barcode is not readable. Camera should response with error after a given time to read the barcode.

sibelius, 2019-03-26:
please share some code or some repro

Thomazella, 2019-03-27:
> I am looking for something to handle when the barcode is not readable. The camera should respond with an error after a given time to read the barcode.

@sibelius you think this is something worth implementing as a feature? Looking for some projects with native ios code to contribute to

sibelius, 2019-03-27:
I'm not sure if this is android or ios related

kunalkp, 2019-03-28:
This is not OS specific. In general we don't get any error when Barcode is not readable if it's blur or any other reason.  It should be a must feature to implement.

closed 2019-06-07 by stale[bot]

andresmtz98, 2019-08-15:
Is there a way to improve the barcode reading in cameras with low resolution (old devices)?