react-native-camera/react-native-camera #2132: null is not an object (evaluating 'CameraManager.Aspect')

AcidNoX, 2019-03-05:
# Bug Report
**To Do First**
- [ Yes ] Did you try latest release?
- [ Yes ] Did you try master?
- [ Yes ] Did you look for existing matching issues?

**Related Modules**

RNCamera

**Platforms**

iOS

**Versions**

- Android:
- iOS: 12.1
- react-native-camera: 1.12.0
- react-native: 0.58.6
- react: 16.6.3
- react-navigation: 3.3.2

**Description/Current Behaviour**

Importing RNCamera gives the error "null is not an object (evaluating 'CameraManager.Aspect')"

I've tried the react-native link and also the manual process of adding rnc using the link binary with libraries but getting no where with it.

I'm not even using "RNCamera" anywhere in the component yet, I've just added it via yarn and done the link / manual install.

Any help would be greatly appreciated!

**Expected Behaviour**

Used react-native init so not using Expo.

**Additionals**

[image]

AcidNoX, 2019-03-05:
To add to this, I've followed all of the non-expo specific instructions as described in the below link and restarted my Mac to boot. Still having no luck getting this working. 

https://forums.expo.io/t/how-to-clear-the-react-native-packager/1352

I have managed to get it not erroring in a totally clean new app but i *should* be able to install it in to an existing app, right?

AcidNoX, 2019-03-06:
So eventually, after spending an entire day and evening on this, i was able to get it working by deleting the ios/build directory.

closed 2019-03-06 by AcidNoX

Synx94, 2019-03-27:
Hey- I have same issue on an existing project- do you mean you completely delete the ios/build directory or the RNcamera files from within them?

AcidNoX, 2019-03-27:
Hi, I deleted the entire ios/build directory

Synx94, 2019-03-27:
I've now done this and still giving me same error. Can you think of anything else you have done?

caki0915, 2019-04-01:
I had the same problem. I forgot to run `pod install` in the `ios` folder.
After that it worked

AcidNoX, 2019-04-01:
Not sure about @Synx94 but I was using react-native link so no pods involved. Thanks for your input  though 👍

c4xuange, 2019-04-05:
I resolved this issue by rebuilding the app

mengheangrat, 2019-06-14:
I resolved this issue by update to `"react-native-camera": "^2.10.2"`