react-native-camera/react-native-camera #2133: Adding burst capture for Android

SirNeuman, 2019-03-05:
# Feature-Request

**Describe the Feature**
I would like to be able to utilize burst capture using the camera on Android. 

**Possible Implementations**
There is a method in the camera2 API in the CameraCaptureSession class that should be able to accomplish this.
https://developer.android.com/reference/android/hardware/camera2/CameraCaptureSession.html#captureBurst(java.util.List%3Candroid.hardware.camera2.CaptureRequest%3E,%20android.hardware.camera2.CameraCaptureSession.CaptureCallback,%20android.os.Handler)

**Related Issues**
I couldn't find any other related issues or requests for this feature, but I would like this feature for my own application. I at least plan on creating a fork and developing it for my own fork on Android. I was wondering if I were to submit a PR upstream whether or not this feature could be accepted and if there are any recommendations for implementation to get this to be accepted into the upstream repo.

closed 2019-06-07 by stale[bot]

aprilmintacpineda, 2020-07-27:
This is a good feature to add, how come it got ignored?

visheshagomic, 2021-09-13:
Yes, I also want to open this issue