react-native-camera/react-native-camera #2134: on

isaaclem, 2019-03-06:
# Bug Report
**To Do First**
- [x] Did you try latest release?
- [ ] Did you try master?
- [x] Did you look for existing matching issues?

**Related Modules**
RNCamera

**Platforms**
Android

**Versions**
- Android: 9
- react-native-camera: 1.12.0
- react-native: ^0.57.5
- react: 16.6.1

**Description/Current Behaviour**
```js
onBarCodeRead = ({ data }) => {
...
};

<RNCamera
  onBarCodeRead={onBarCodeRead}
  captureAudio={false}
>
...
</RNCamera>
```
onBarCodeRead is not firing at all on Android Samsung S9. The same code is working fine on iOS

**Expected Behaviour**
A simple calling `onBarCodeRead` is good enough

It does show the camera, but simply not detecting barcode/qrcode and invoke `onBarCodeRead`

isaaclem, 2019-03-06:
The reported issue was with debug mode. I've just tested on released build it's working fine.

wesleyguirra, 2019-04-17:
same issue

sibelius, 2019-04-17:
Try with latest version

closed 2019-04-18 by isaaclem