react-native-camera/react-native-camera #2137: Is there any way to compress the recorded video

HengCC, 2019-03-11:
# Question

**To Do First**
- [x] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [x] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [x] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**

Is there any way to compress the recorded video, I tried to record a 30-second video, the recording parameter is `RNCamera.Constants.VideoQuality.480p` . But on the Android platform, the size is 18M, and at the same time there is 12M size on the IOS, I think it should Can control below 8M, is there a way?  Thanks~

**Tags**

compress video
> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

HengCC, 2019-03-11:
I set `videoBitrate ` to `2*1024*1024`, now  on the Android platform, the size is 6.6M, This is what I want, But looking at the API, the parameter seems to only work on Android.

medicmilos, 2019-03-12:
> This is what I want, But looking at the API, the parameter seems to only work on Android.

I think that **iOS devices** compresses videos automatically after picking it from library.

HengCC, 2019-03-13:
@medicmilos this  [https://github.com/react-native-community/react-native-camera/issues/687](issue) was originally raised for this question. But it's just Android. The author does not seem to be developing IOS.So I guess the implementation of this parameter will only working on Android.

closed 2019-06-07 by stale[bot]