react-native-camera/react-native-camera #2138: Cannot choose between the following variants of project :react-native-camera:

medicmilos, 2019-03-11:
# Bug Report
**To Do First**
- [Yes ] Did you try latest release?
- [Yes] Did you try master?
- [Yes] Did you look for existing matching issues?

**Related Modules**
RNCamera

**Platforms**
Android

**Versions**

- Android: 27
- react-native-camera: v2.0.0
- react-native: 0.58.5
- react: 16.6.3
- react-navigation: 3.1.2

**Description/Current Behaviour**

Sincle today's release (v2.0.0), can't run project anymore. I have tried v1.13.1 and it works fine.

**Expected Behaviour**

**Steps to Reproduce**

**Additionals**

[image]

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

sibelius, 2019-03-11:
can you check both `basic` and `mlkit` example?

are you using face recognition?

medicmilos, 2019-03-11:
Of Course i have checked the examples. Everything was working (same code on both releases).
It's nothing major on my side, I can work with previous version. Just wanted to notice that something might be wrong.
I'm not using face recognition. I'm using basic functionalities of RNCamera, as following:

[image]

sibelius, 2019-03-11:
does basic example compile in your machine?

wimva, 2019-03-11:
Exactly the same issue here ✋

dov11, 2019-03-11:
@medicmilos, @wimva , @MeisterTea , have you added the following to `android/app/build.gradle`?:
```
android {
  ...
  defaultConfig {
    ...
    missingDimensionStrategy 'react-native-camera', 'general' <-- insert this line
  }
}
```
Step 4 [here](https://github.com/react-native-community/react-native-camera#android)

Time-coding, 2019-03-12:
@dov11 Thanks

MeisterTea, 2019-03-12:
> @medicmilos, @wimva , @MeisterTea , have you added the following to `android/app/build.gradle`?:
>
> ```
> android {
>   ...
>   defaultConfig {
>     ...
>     missingDimensionStrategy 'react-native-camera', [...]

Well, it's in the "Manual" section. It should be added automatically right ?

sibelius, 2019-03-12:
Follow this https://github.com/react-native-community/react-native-camera/issues/2138#issuecomment-471669956

closed 2019-03-12 by sibelius

dov11, 2019-03-12:
@MeisterTea,
This line is not part of automatic installation, it allows users to choose between the flavours of the android part of the library, by either adding `missingDimensionStrategy 'react-native-camera', 'general'` or `missingDimensionStrategy 'react-native-camera', 'mlkit'`. This is done so that users who do not need Firebase MLKit-based features, such as text/face recognition, are not forced to set up a firebase project. 

Currently, if you select `general` flavour you can use text/face recognition based on google mobile vision, but it is being deprecated, and probably we might want to move it to a different flavour - this should allow us to remove `"com.google.android.gms:play-services-vision:$googlePlayServicesVisionVersion"` dependency and make life easier for users that do not need this features at all.

The need to add one of the lines above is a braking change and the reason why there is a bump to 2.0. In hindsight, I think we should have added migration docs prior to release.

brunsy, 2019-03-18:
In case anyone else had issues with the above.

The missingDimensionStrategy step didn't work for me when building via fastlane so ended up using [filter-variants](https://developer.android.com/studio/build/build-variants#filter-variants) to strip out the mlkit stuff

chriszwickerocteris, 2019-03-28:
Just my two cents: I would suggest adding the line mentioned automatically and putting instructions how to change it in the docs. Many users will give up easily if there's no standard that works out of the box.

ehorodyski, 2019-04-19:
This should be part of the docs for new installs/linking. Thanks for the help all.

rphlmr, 2019-04-23:
> @medicmilos, @wimva , @MeisterTea , have you added the following to `android/app/build.gradle`?:
>
> ```
> android {
>   ...
>   defaultConfig {
>     ...
>     missingDimensionStrategy 'react-native-camera', [...]

When I do that it works but breaks my react-native-navigation module :( 

`missingDimensionStrategy "react-native-camera", "general", "RNN.reactNativeVersion", "reactNative57_5"
`

witalobenicio, 2019-06-11:
@rphlmr +1 for the tip about RNN and for using RNN. 😄

hu-qi, 2019-06-24:
> @medicmilos, @wimva , @MeisterTea , have you added the following to `android/app/build.gradle`?:
>
> ```
> android {
>   ...
>   defaultConfig {
>     ...
>     missingDimensionStrategy 'react-native-camera', [...]

Thanks， It works for me.

frontendjorge, 2019-07-18:
> @MeisterTea,
> This line is not part of automatic installation, it allows users to choose between the flavours of the android part of the library, by either adding `missingDimensionStrategy 'react-native-camera', 'general'` or `missingDimensionStrategy 'react-native-camera', 'mlkit'`. This is done so that users who do not need Firebase MLKit-based features, such as text/face recognition, are not forced to set up a firebase project.
> 
> Currently, if you select `general` flavour you can use text/face recognition based on google mobile vision, but it is being deprecated, and probably we might want to move it to a different flavour - this should allow us to remove `"com.google.android.gms:play-services-vision:$googlePlayServicesVisionVersion"` dependency and make life easier for users that do not need these features at all.
> 
> The need to add one of the lines above is a breaking change and the reason why there is a bump to 2.0. In hindsight, I think we should have added migration docs prior to release.

After this it's possible you need to add in android//app/build.gradle  compile 'com.android.support:multidex:1.0.3' because the libraries has the number of method references in a .dex file (cannot exceed 64K)

vangthachdinh, 2019-08-12:
> @medicmilos, @wimva , @MeisterTea , have you added the following to `android/app/build.gradle`?:
>
> ```
> android {
>   ...
>   defaultConfig {
>     ...
>     missingDimensionStrategy 'react-native-camera', [...]

It's works for me. Thanks !!!

gopinav, 2019-08-13:
As soon as I add `missingDimensionStrategy 'react-native-camera', 'general'`,
react-native-gesture-handler starts throwing error for me.
Please let me know how to fix this

`Task :react-native-gesture-handler:compileDebugJavaWithJavac FAILED
/Users/vishwas/Documents/react_native_app/node_modules/react-native-gesture-handler/android/src/main/java/com/swmansion/gesturehandler/react/RNGestureHandlerEvent.java:
3: error: package android.support.v4.util does not exist
import android.support.v4.util.Pools;
                              
/Users/vishwas/Documents/react_native_app/node_modules/react-native-gesture-handler/android/src/main/java/com/swmansion/gesturehandler/react/RNGestureHandlerEvent.java:19: error: package Pools does not exist
  private static final Pools.SynchronizedPool<RNGestureHandlerEvent> EVENTS_POOL =
                            
/Users/vishwas/Documents/react_native_app/node_modules/react-native-gesture-handler/android/src/main/java/com/swmansion/gesturehandler/react/RNGestureHandlerStateChangeEvent.java:3: error: package android.support.v4.util does not exist
import android.support.v4.util.Pools;
                              
/Users/vishwas/Documents/react_native_app/node_modules/react-native-gesture-handler/android/src/main/java/com/swmansion/gesturehandler/react/RNGestureHandlerStateChangeEvent.java:19: error: package Pools does not exist
  private static final Pools.SynchronizedPool<RNGestureHandlerStateChangeEvent> EVENTS_POOL =
                            
/Users/vishwas/Documents/react_native_app/node_modules/react-native-gesture-handler/android/src/main/java/com/swmansion/gesturehandler/react/RNGestureHandlerEvent.java:20: error: package Pools does not exist
          new Pools.SynchronizedPool<>(TOUCH_EVENTS_POOL_SIZE);
                   
/Users/vishwas/Documents/react_native_app/node_modules/react-native-gesture-handler/android/src/main/java/com/swmansion/gesturehandler/react/RNGestureHandlerStateChangeEvent.java:20: error: package Pools does not exist
          new Pools.SynchronizedPool<>(TOUCH_EVENTS_POOL_SIZE);
                   
Note: /Users/vishwas/Documents/react_native_app/node_modules/react-native-gesture-handler/android/src/main/java/com/swmansion/gesturehandler/react/RNGestureHandlerButtonViewManager.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
6 errors`

gustperz, 2019-08-31:
same error @gopinav

devamaz, 2019-09-02:
Have sane error @gopinav and @gustperz  have any been able to fix this? @dov11 Please help

gopinav, 2019-09-03:
@gustperz  @devamaz  - This happened with me because the latest react-native-camera version is for RN 0.60 and above. Install RNCamera version 2.11.2 and you should be ok. (Or upgrade your project to 0.60)

GunnarAK, 2019-10-08:
Installation documentation fixing this issue can now be found here:
https://react-native-community.github.io/react-native-camera/docs/installation#android 

`inside defaultConfig block insert either:`
```
android {
  ...
  defaultConfig {
    ...
    missingDimensionStrategy 'react-native-camera', 'general' <-- insert this line
  }
}
```

raimonkh, 2020-02-12:
This is an issue when upgrading to RN > 0.61 and react-native-camera 3.17
I have this code that you suggest and I still get the error.

```
A problem occurred configuring project ':react-native-camera'.
> Flavor 'general' has unknown dimension 'react-native-camera'.
```

I'm running out of ideas.

GunnarAK, 2020-02-13:
Have you followed the instructions here?
https://react-native-community.github.io/react-native-camera/docs/installation#android

raimonkh, 2020-02-13:
Yes  @GunnarAK to the point. The fact is that it doesn't matter if I add the `missingDimensionStrategy`or not. 
I've read all gradle docs about it (because we do have 2 `productFlavors` but still not working. Error logs don't give anything. I wonder if having different configuration and flavours might do something...
I'm on RN 0.61.5, JDK 1.8, classpath 3.5.3, gradle-5.4.1... not sure if it's lib related or not...

raimonkh, 2020-02-13:
Ok I've grilled down the problem.
we have more flavours of the app (2) and 1 dimension.
We also have subprojects configured, that seems to not be liked by the way you have the camera setup... If I add dimension and flavour to the app it doesn't work

maxkomarychev, 2020-04-10:
I managed to get multi-flavor app compiling by moving `flavorDimensions "default"` from android/build.gradle:` down to `android/app/build.gradle`

so the final scheme looks like this:

android/build.gradle:
```gradle
subprojects {
    afterEvaluate {project ->
        if (project.hasProperty("android")) {
            android {
                 ....
                flavorDimensions "default"  // <= remove this!
            }
        }
    }
```

android/app/build.gradle
```
android {
  defaultConfig {
    missingDimensionStrategy '....'
  }

  
    flavorDimensions "default" // <= add this! right before custom flavors
    productFlavors {
        myflavor1 {}
        myflavor2 {
        }
    }
}

arwysyah, 2020-11-26:
hei i still getting same issue, i already followed the instructions

Malik-Usman17, 2023-03-13:
> missingDimensionStrategy 'react-native-camera', 'general'

Works for me, thanks