react-native-camera/react-native-camera #2140: set display orientation failed

dimlksin2, 2019-03-12:
android_version: 8.1.0
java.lang.RuntimeException: set display orientation failed
	at android.hardware.Camera.setDisplayOrientation(Native Method)
	at com.lwansbrough.RCTCamera.RCTCamera.adjustPreviewLayout(RCTCamera.java:427)
	at com.lwansbrough.RCTCamera.RCTCamera.setActualDeviceOrientation(RCTCamera.java:211)
	at com.lwansbrough.RCTCamera.RCTCameraView.setActualDeviceOrientation(RCTCameraView.java:160)
	at com.lwansbrough.RCTCamera.RCTCameraView.access$100(RCTCameraView.java:16)
	at com.lwansbrough.RCTCamera.RCTCameraView$1.onOrientationChanged(RCTCameraView.java:37)
	at android.view.OrientationEventListener$SensorEventListenerImpl.onSensorChanged(OrientationEventListener.java:143)
	at android.hardware.SystemSensorManager$SensorEventQueue.dispatchSensorEvent(SystemSensorManager.java:830)
	at android.os.MessageQueue.nativePollOnce(Native Method)
	at android.os.MessageQueue.next(MessageQueue.java:366)
	at android.os.Looper.loop(Looper.java:144)
	at android.app.ActivityThread.main(ActivityThread.java:6861)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:450)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:936)

dimlksin2, 2019-03-12:
Are you going to fix the error? This problem is urgent for me, a year has passed

dimlksin2, 2019-03-12:
or tell me how to get around your mistake

sibelius, 2019-03-12:
@dimlksin2 feel free to give an issuehunt bounty or send a fix for this

closed 2019-04-18 by sibelius