react-native-camera/react-native-camera #2141: Expose text property when recognizing text using mlkit

Lythenas, 2019-03-13:
# Feature-Request

**Describe the Feature**

In addition to the text blocks MlKit also exposes a `getText()` method that returns the complete recognized text. See https://firebase.google.com/docs/reference/android/com/google/firebase/ml/vision/text/FirebaseVisionText

To me this would be useful so I don't have to stitch together the text myself.

**Possible Implementations**

Change `TextRecognizerAsyncTaskDelegate.onTextRecognized` and `TextRecognizedEvent` to take an additional parameter containing the text. And change `TextRecognizedEvent.createEvent` to also serialize the text.

I can create a pull request if you want to add that feature.

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this feature to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

sibelius, 2019-03-13:
can you send a pull request improving this

Lythenas, 2019-03-13:
I will give it a try

closed 2019-06-02 by Lythenas