react-native-camera/react-native-camera #2146: Is it possible to render camera in a circle?

williamgurzoni, 2019-03-14:
I've tried all this options, with no success:

- RNCamera inside a circular SVG
- RNCamera with radius border
- RNCamera inside circular View

madhumatip, 2019-03-15:
yes ,
you can apply style to RNCamera 
for eg. camera: {
        height: hp('40%'),
        width: hp('40%'),
        borderRadius: hp('20%'), 
    }

sibelius, 2019-03-15:
please add this to the docs

williamgurzoni, 2019-03-15:
Hey @madhumatip, I tried this, but no success. Am I doing something wrong?

`<RNCamera
          style={{ height: hp('40%'), width: hp('40%'), borderRadius: hp('20%'), }}
...`

madhumatip, 2019-03-18:
can please share your code because it works for me

darekg11, 2019-03-20:
We are also doing this in `circle` shape with following styles (working on both IOS and Android):
```
            <View style={ styles.preview }>
              <RNCamera
                ref={ cam => {
                  this.camera = cam
                } }
                style={ styles.camera }
                type={ RNCamera.Constants.Type.front }
                flashMode={ RNCamera.Constants.FlashMode.on }
                ratio={ '16:9' }
              />
            </View>
```

```
  preview: {
    height: 240,
    width: 240,
    borderRadius: 120,
    overflow: 'hidden'
  },
  camera: {
    height: 240,
    width: 240
}
```
`React-Native-Camera`: `1.1.2`

Let me know if this would give you results closer to what you want to achieve, I can also fork the repo and do it on the demo app.

liangguohun, 2019-04-14:
i want to know if use RNCamera.Constants.Type.front the photo result will yours' like the camera show or change direct

MateusAndrade, 2019-04-27:
> i want to know if use RNCamera.Constants.Type.front the photo result will yours' like the camera show or change direct

If you are asking about the picture orientation, i think this will change the direction. You can found more about this on the instructions provided on the DOCS.

closed 2019-07-03 by stale[bot]