react-native-camera/react-native-camera #2150: Could not resolve project :react-native-camera. on Android

HugoSanchez, 2019-03-17:
Hi, 

After running `npm install --save react-native-camera`  I get the following error when trying I run the project with `react-native run-android`:

`* What went wrong:
Could not determine the dependencies of task ':app:mergeDebugAssets'.
> Could not resolve all task dependencies for configuration ':app:debugRuntimeClasspath'.
   > Could not resolve project :react-native-camera.
     Required by:
         project :app
      > Cannot choose between the following configurations of project :react-native-camera:
          - generalDebugRuntimeElements
          - mlkitDebugRuntimeElements
        All of them match the consumer attributes:
          - Configuration 'generalDebugRuntimeElements':
              - Required com.android.build.api.attributes.BuildTypeAttr 'debug' and found compatible value 'debug'.
              - Found com.android.build.api.attributes.VariantAttr 'generalDebug' but wasn't required.
              - Required com.android.build.gradle.internal.dependency.AndroidTypeAttr 'Aar' and found compatible value 'Aar'.
              - Required org.gradle.usage 'java-runtime' and found compatible value 'java-runtime'.
              - Found react-native-camera 'general' but wasn't required.
          - Configuration 'mlkitDebugRuntimeElements':
              - Required com.android.build.api.attributes.BuildTypeAttr 'debug' and found compatible value 'debug'.
              - Found com.android.build.api.attributes.VariantAttr 'mlkitDebug' but wasn't required.
              - Required com.android.build.gradle.internal.dependency.AndroidTypeAttr 'Aar' and found compatible value 'Aar'.
              - Required org.gradle.usage 'java-runtime' and found compatible value 'java-runtime'.
              - Found react-native-camera 'mlkit' but wasn't required.`

I have no idea what is wrong.

Versions:
react-native-cli: 2.0.1
react-native: 0.58.6
Android 7 

Can anyone provide some help please?

Thanks!

sibelius, 2019-03-17:
Follow migration guide to v2

amed, 2019-03-18:
After using AndroidStudio, error message is shown as following: 
`Unable to resolve dependency for ':react-native-camera@debug/compileClasspath': Could not resolve com.android.support:support-compat:27.1.1.`

dov11, 2019-03-18:
@HugoSanchez, have you added to your `android/app/build.gradle` either 
```
missingDimensionStrategy 'react-native-camera', 'general'
``` 
or 
```
missingDimensionStrategy 'react-native-camera', 'mlkit'
```
following [this](https://github.com/react-native-community/react-native-camera/blob/master/docs/migrationV2.md#android)?

MrDatastorage, 2019-03-20:
I have the same issue
```   > Could not resolve project :react-native-camera.
     Required by:
         project :app
      > Cannot choose between the following variants of project :react-native-camera:
          - generalDebugRuntimeElements
          - mlkitDebugRuntimeElements
        All of them match the consumer attributes:
          - Variant 'generalDebugRuntimeElements':
              - Required com.android.build.api.attributes.BuildTypeAttr 'debug' and found compatible value 'debug'.
              - Found com.android.build.api.attributes.VariantAttr 'generalDebug' but wasn't required.
              - Required com.android.build.gradle.internal.dependency.AndroidTypeAttr 'Aar' and found compatible value 'Aar'.
              - Required org.gradle.usage 'java-runtime' and found compatible value 'java-runtime'.
              - Found react-native-camera 'general' but wasn't required.
          - Variant 'mlkitDebugRuntimeElements':
              - Required com.android.build.api.attributes.BuildTypeAttr 'debug' and found compatible value 'debug'.
              - Found com.android.build.api.attributes.VariantAttr 'mlkitDebug' but wasn't required.
              - Required com.android.build.gradle.internal.dependency.AndroidTypeAttr 'Aar' and found compatible value 'Aar'.
              - Required org.gradle.usage 'java-runtime' and found compatible value 'java-runtime'.
              - Found react-native-camera 'mlkit' but wasn't required.```

MrDatastorage, 2019-03-20:
> @HugoSanchez, have you added to your `android/app/build.gradle` either
>
> ```
> missingDimensionStrategy 'react-native-camera', 'general'
> ```
>
> or
>
> ```
> missingDimensionStrategy 'react-native-camera', [...]

Didn't resolve the issue for me.

manelsanz, 2019-03-22:
I have the same problems after update RN to 0.59.1 and Gradle 4.4. I think that react-native-camera doesn't suport API 28, or something like that...

smeekmeister, 2019-03-22:
I have the same build problems

manelsanz, 2019-03-22:
I solved my build errors uninstalling react-native-camera, setting classpath 3.3.0 and Gradle 4.10.1, and then reinstalling the package following github page instructions.
I hope that helps someone..

closed 2019-03-23 by sibelius

tadeumx1, 2019-03-24:
For those who have this problem, the solution that I found is setting classpath 3.3.0 and Gradle 4.10.1 as already mentioned, to do this I changed the class path in ```android/build.gradle```in ```dependencies```tag now is like this ``` classpath 'com.android.tools.build:gradle:3.3.0' ``` and changed the gradle version in ``` android/wrapper/gradle-wrapper.properties``` to this 

``` distributionUrl=https\://services.gradle.org/distributions/gradle-4.10.1-all.zip ``` 

and in the ```android/app/build.gradle ``` I added this 

``` missingDimensionStrategy 'react-native-camera', 'general' ``` in ```defaultConfig ``` 

And i also you added these two lines 

``` 
maven { url "https://jitpack.io" } 
maven { url "https://maven.google.com" } 
``` 
in repositories tag of ```android/build.gradle ```

And then I could use the camera normally

tadeumx1, 2019-03-24:
For those who have this problem, the solution that I found is setting classpath 3.3.0 and Gradle 4.10.1 as already mentioned, to do this I changed the class path in ```android/build.gradle```in ```dependencies```tag now is like this ``` classpath 'com.android.tools.build:gradle:3.3.0' ``` and changed the gradle version in ``` android/wrapper/gradle-wrapper.properties``` to this 

``` distributionUrl=https\://services.gradle.org/distributions/gradle-4.10.1-all.zip ``` 

and in the ```android/app/build.gradle ``` I added this 

``` missingDimensionStrategy 'react-native-camera', 'general' ``` in ```defaultConfig ``` 

And i also added these two lines 

``` 
maven { url "https://jitpack.io" } 
maven { url "https://maven.google.com" } 
``` 
in repositories tag of ```android/build.gradle ```

And then I could use the camera normally

tadeumx1, 2019-03-24:
For those who have this problem, the solution that I found is setting classpath 3.3.0 and Gradle 4.10.1 as already mentioned, to do this I changed the class path in ```android/build.gradle```in ```dependencies```tag now is like this ``` classpath 'com.android.tools.build:gradle:3.3.0' ``` and changed the gradle version in ``` android/wrapper/gradle-wrapper.properties``` to this 

``` distributionUrl=https\://services.gradle.org/distributions/gradle-4.10.1-all.zip ``` 

and in the ```android/app/build.gradle``` I added this 

``` missingDimensionStrategy 'react-native-camera', 'general' ``` in ```defaultConfig``` 

And i also added these two lines 

``` 
maven { url "https://jitpack.io" } 
maven { url "https://maven.google.com" } 
``` 
in repositories tag of ```android/build.gradle ```

And then I could use the camera normally

gabrielclima, 2019-03-30:
@tadeumx1 worked well to me, thank you 🤘

pjunior94, 2019-04-01:
> For those who have this problem, the solution that I found is setting classpath 3.3.0 and Gradle 4.10.1 as already mentioned, to do this I changed the class path in `android/build.gradle`in [...]

I took all day long to find this! Thank you man!

MadhaviJayasinghe, 2019-04-03:
@tadeumx1 worked for me. Thanks a lot.

HugoSanchez, 2019-04-12:
Thanks @tadeumx1 !!

Rishee123, 2019-04-12:
i have uninstalled the 'react-native-camera' package still i am getting the following error

FAILURE: Build failed with an exception.

* What went wrong:
Could not determine the dependencies of task ':app:preDebugBuild'.
> Could not resolve all task dependencies for configuration ':app:debugRuntimeClasspath'.
   > Could not resolve project :react-native-camera.
     Required by:
         project :app
      > Unable to find a matching configuration of project :react-native-camera: None of the consumable configurations have attributes.

LiuXiangX, 2019-04-16:
@tadeumx1  Thanks

mrcarjul, 2019-04-16:
@tadeumx1 thanks bro you da best

manelsanz, 2019-04-17:
OMG I explained first the same as  @tadeumx1 and everybody is thanking him. :(

tadeumx1, 2019-04-17:
You didnt explained @manelsanz just said the things, and i did many changes to use the camera hahahaha

manelsanz, 2019-04-17:
Did you many changes of what? You said LITERALLY that says **README**. And that is what I indicated on my comment, before than you.

**README**:

> inside defaultConfig block insert either: 
```
android {
  ...
  defaultConfig {
    ...
    missingDimensionStrategy 'react-native-camera', 'general' <-- insert this line
  }
}

```

> Add jitpack to android/build.gradle
```
allprojects {
    repositories {
        maven { url "https://jitpack.io" }
        maven { url "https://maven.google.com" }
    }
}
```

tadeumx1, 2019-04-17:
Yes exactly that instead of going to README they just saw my comment, and i said in my comment that these changes had already been mentioned

hewking, 2019-04-25:
> For those who have this problem, the solution that I found is setting classpath 3.3.0 and Gradle 4.10.1 as already mentioned, to do this I changed the class path in `android/build.gradle`in [...]

thank you ,i solve it

saraMarzouk, 2019-07-01:
tadeumx1

 thank you for yur time

Ronin20, 2019-07-10:
Thank you @tadeumx1

eddskt, 2019-07-24:
> For those who have this problem, the solution that I found is setting classpath 3.3.0 and Gradle 4.10.1 as already mentioned, to do this I changed the class path in `android/build.gradle` in `dependencies` tag now is like this `classpath 'com.android.tools.build:gradle:3.3.0'` and changed the gradle version in `android/gradle/wrapper/gradle-wrapper.properties` to this
> 
> `distributionUrl=https\://services.gradle.org/distributions/gradle-4.10.1-all.zip`
> 
> and in the `android/app/build.gradle` I added this
> 
> `missingDimensionStrategy 'react-native-camera', 'general'` in `defaultConfig`
> 
> And i also added these two lines
> 
> ```
> maven { url "https://jitpack.io" } 
> maven { url "https://maven.google.com" } 
> ```
> 
> in repositories tag of `android/build.gradle `
> 
> And then I could use the camera normally

Thanks, one of the more important steps was add missingDimensionStrategy on build.gradle app module and add Maven Lines in build.gradle on project level

sapjax, 2019-08-27:
Doesn't work for me:

> A problem occurred configuring project ':react-native-camera'.
> \> Flavor 'general' has unknown dimension 'react-native-camera'.

TakaKeiji, 2019-09-26:
> Thanks, one of the more important steps was add missingDimensionStrategy on build.gradle app module and add Maven Lines in build.gradle on project level

Thanks this solves my problem

Topdeveloperpcp, 2019-10-06:
I have got error.
In app/build.gradle.
I put the code "missingDimensionStrategy 'react-native-camera', 'general'".
[image]
I have fixed

hzburki, 2019-10-11:
# Error
```
FAILURE: Build failed with an exception.

* What went wrong:
Could not determine the dependencies of task ':react-native-camera:compileMlkitReleaseAidl'.
> Could not resolve all task dependencies for configuration ':react-native-camera:mlkitReleaseCompileClasspath'.
   > Could not find com.google.firebase:firebase-ml-vision:12.0.1.
     Required by:
         project :react-native-camera
   > Could not find com.google.firebase:firebase-ml-vision-face-model:12.0.1.
     Required by:
         project :react-native-camera
```

### android/app/build.gradle
```
android {
     ...
    defaultConfig {
      applicationId "com.numu.app"
      minSdkVersion rootProject.ext.minSdkVersion
      targetSdkVersion 28
      versionCode 89
      versionName "4.4.2"
      missingDimensionStrategy 'react-native-camera', 'general'
    }
    ...
}
```

### android/build.gradle
```
allprojects {
    ...
    repositories {
        mavenLocal()
        jcenter()
        maven { url "https://www.jitpack.io" }
        maven { url "https://maven.google.com" }
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url "$rootDir/../node_modules/react-native/android"
        }
        google()
    }
}
```

fahidnasir, 2019-10-23:
@Topdeveloperpcp solution worked from me. 
I didn't change anything in Podfile, info.plist and <project>/build.gradle files.

I had:

- React Native 0.61.2
- React-Native-Camera v3.8.0
- In graddle-wrapper.properties => `distributionUrl=https\://services.gradle.org/distributions/gradle-5.5-all.zip`
- In <project>/build.gradle `classpath("com.android.tools.build:gradle:3.4.2")`

In <project>/app/build.gradle.

I copied this in <project>/app/build.gradle inside `android => defaultConfig`
`missingDimensionStrategy 'react-native-camera', 'general'`

popstheman, 2019-11-09:
@Topdeveloperpcp solution worked for me as well. 

Just added `missingDimensionStrategy 'react-native-camera', 'general'` in defaultConfig section of /app/build.gradle.

siddhi-sandy, 2019-11-21:
> # Error
> ```
> FAILURE: Build failed with an exception.
> 
> * What went wrong:
> Could not determine the dependencies of task ':react-native-camera:compileMlkitReleaseAidl'.
> > Could not resolve all task dependencies for configuration ':react-native-camera:mlkitReleaseCompileClasspath'.
>    > Could not find com.google.firebase:firebase-ml-vision:12.0.1.
>      Required by:
>          project :react-native-camera
>    > Could not find com.google.firebase:firebase-ml-vision-face-model:12.0.1.
>      Required by:
>          project :react-native-camera
> ```
> 
> ### android/app/build.gradle
> ```
> android {
>      ...
>     defaultConfig {
>       applicationId "com.xxx.app"
>       minSdkVersion rootProject.ext.minSdkVersion
>       targetSdkVersion 28
>       versionCode 89
>       versionName "4.4.2"
>       missingDimensionStrategy 'react-native-camera', 'general'
>     }
>     ...
> }
> ```
> 
> ### android/build.gradle
> ```
> allprojects {
>     ...
>     repositories {
>         mavenLocal()
>         jcenter()
>         maven { url "https://www.jitpack.io" }
>         maven { url "https://maven.google.com" }
>         maven {
>             // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
>             url "$rootDir/../node_modules/react-native/android"
>         }
>         google()
>     }
> }
> ```

Having same issue, anyone know how to fix.
PS: I am using React Native @0.59.10.

siddhi-sandy, 2019-11-21:
* What went wrong:
Could not determine the dependencies of task ':react-native-camera:compileGeneralDebugAidl'.
> Could not resolve all task dependencies for configuration ':react-native-camera:generalDebugCompileClasspath'.
> Could not find com.google.android.gms:play-services-vision:16.1.0.
Required by:
project :react-native-camera

ydv0121, 2019-11-22:
> * What went wrong:
>   Could not determine the dependencies of task ':react-native-camera:compileGeneralDebugAidl'.

i also getting this.any solution? react native camera 2.6.0

ririlya94, 2019-11-22:
> @HugoSanchez, have you added to your `android/app/build.gradle` either
>
> ```
> missingDimensionStrategy 'react-native-camera', 'general'
> ```
>
> or
>
> ```
> missingDimensionStrategy 'react-native-camera', [...]

For people who have this problem, don't use the master branch. I scratch my head alot finding what's the problem even though i follow this guide and other who comment it. For me I use react-native-camera:"^3.11.1" for me to work!

rchouhan170590, 2019-11-25:
You’ll also need to set the dimension strategy in android/app/build.gradle

defaultConfig {
         applicationId "com.cameraexample"
         minSdkVersion rootProject.ext.minSdkVersion
+        missingDimensionStrategy 'react-native-camera', 'general'
         targetSdkVersion rootProject.ext.targetSdkVersion
         versionCode 1
         versionName "1.0"
}

hanaechahid, 2019-12-09:
> @HugoSanchez, have you added to your `android/app/build.gradle` either
>
> ```
> missingDimensionStrategy 'react-native-camera', 'general'
> ```
>
> or
>
> ```
> missingDimensionStrategy 'react-native-camera', [...]

@dov11 thank you for your response but always get the error: **> Task :react-native-camera:javaPreCompileDebug FAILED**

abubaker-hmd, 2020-01-17:
> For those who have this problem, the solution that I found is setting classpath 3.3.0 and Gradle 4.10.1 as already mentioned, to do this I changed the class path in `android/build.gradle` in [...]

Only adding the  missingDimensionStrategy can fix it for me
Thank you

Dschell-w, 2020-02-04:
> Did you many changes of what? You said LITERALLY that says **README**, and the versions of Classpath and Gradle. And that is what I indicated on my comment, before than you.
> 
> **README**:
> 
> > inside defaultConfig block insert either:
> 
> ```
> android {
>   ...
>   defaultConfig {
>     ...
>     missingDimensionStrategy 'react-native-camera', 'general' <-- insert this line
>   }
> }
> ```
> 
> > Add jitpack to android/build.gradle
> 
> ```
> allprojects {
>     repositories {
>         maven { url "https://jitpack.io" }
>         maven { url "https://maven.google.com" }
>     }
> }
> ```

Your help (@manelsanz ) and the help of (@Topdeveloperpcp) has help me. Thanks you!!! I could understand this easy.

carlosstenzel, 2020-03-16:
> @HugoSanchez, have you added to your `android/app/build.gradle` either
>
> ```
> missingDimensionStrategy 'react-native-camera', 'general'
> ```
>
> or
>
> ```
> missingDimensionStrategy 'react-native-camera', [...]

solved for me

shahan-ayyub, 2020-04-21:
> @HugoSanchez, have you added to your `android/app/build.gradle` either
>
> ```
> missingDimensionStrategy 'react-native-camera', 'general'
> ```
>
> or
>
> ```
> missingDimensionStrategy 'react-native-camera', [...]

Followed this instruction but had to rollback from version:
"react-native-camera": "^3.23.1",  // This was latest when issue faced
to:
"react-native-camera": "3.22.1",

Jp3rd, 2020-05-26:
> > For those who have this problem, the solution that I found is setting classpath 3.3.0 and Gradle 4.10.1 as already mentioned, to do this I changed the class path in `android/build.gradle` in `dependencies` tag now is like this `classpath 'com.android.tools.build:gradle:3.3.0'` and changed the gradle version in `android/gradle/wrapper/gradle-wrapper.properties` to this
> > `distributionUrl=https\://services.gradle.org/distributions/gradle-4.10.1-all.zip`
> > and in the `android/app/build.gradle` I added this
> > `missingDimensionStrategy 'react-native-camera', 'general'` in `defaultConfig`
> > And i also added these two lines
> > ```
> > maven { url "https://jitpack.io" } 
> > maven { url "https://maven.google.com" } 
> > ```
> > 
> > 
> > in repositories tag of `android/build.gradle `
> > And then I could use the camera normally
> 
> Just by adding the missingDimensionStrategy fixed it for me
> Thank you

So I did all of these, but my previous class path was 3.6.1 and Gradle distributionURL was 5.6.4.
After running the `react-native run-android`, I changed it back, ran the command again and the issue was solved. 

Does anyone know if android/react native has some kind of cache for this? if yes, Is there a way to clear it ?

VSV6, 2020-09-24:
@Chrissutherland211 thanks man it helped me a lot. You said so clearly 👍  :)

Scriptonita, 2020-12-04:
> For those who have this problem, the solution that I found is setting classpath 3.3.0 and Gradle 4.10.1 as already mentioned, to do this I changed the class path in `android/build.gradle` in [...]

After those changes, to get it working I had to add to android/app/build.gradle:

```
buildscript { 
  ext{   
     ... 
    googlePlayServicesVisionVersion = "+" 
  } 
} 
```

sumj25, 2021-03-23:
in the android/app/build.gradle I added this

**missingDimensionStrategy 'react-native-camera', 'general'** and  **multiDexEnabled true** in defaultConfig

And i also added these two lines

**maven { url "https://jitpack.io" } 
maven { url "https://maven.google.com" }** 
in repositories tag of android/build.gradle

And then I could use the camera normally

Signati, 2021-04-06:
> **missingDimensionStrategy 'react-native-camera', 'general'** y **multiDexEnabled true**

esto si me soluciono el problema 

missingDimensionStrategy 'react-native-camera', 'general'
multiDexEnabled true

JaskaranSingh4704, 2021-05-22:
thanks its working

name-q, 2022-08-23:
android/build.gradle
     -        maven { url 'https://maven.google.com' }
     +        maven { url 'https://dl.google.com/dl/android/maven2/' }

maaritv, 2023-02-07:
Are these steps needed every time after installing react-native-camera? Should they be added to installation instructions too?

mohdgulzar, 2023-05-06:
repositories {
        ...
        maven { url "https://jitpack.io" }
        maven { url "https://maven.google.com" }
    }

add in android/build.gradle

and 

defaultConfig {
         . . .
        missingDimensionStrategy 'react-native-camera', 'general' 
    }

add in android/app/biuld.gradle