react-native-camera/react-native-camera #2151: Could not resolve com.google.android.gms:play-services-vision:17.0.2. on Android

ekoooo, 2019-03-18:
# Bug Report
**To Do First**
- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Related Modules**

**Platforms**

Android

**Versions**

- react-native-camera: ^2.0.1
- react-native: 0.57.8
- react: 16.6.3

**Description/Current Behaviour**

FAILURE: Build failed with an exception.
```
FAILURE: Build failed with an exception.

* What went wrong:
Could not resolve all files for configuration ':app:debugRuntimeClasspath'.
> Could not resolve com.google.android.gms:play-services-vision:17.0.2.
  Required by:
      project :app > project :react-native-camera
   > Could not resolve com.google.android.gms:play-services-vision:17.0.2.
      > Could not get resource 'https://dl.google.com/dl/android/maven2/com/google/android/gms/play-services-vision/17.0.2/play-services-vision-17.0.2.pom'.
         > Could not HEAD 'https://dl.google.com/dl/android/maven2/com/google/android/gms/play-services-vision/17.0.2/play-services-vision-17.0.2.pom'.
            > sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
   > Could not resolve com.google.android.gms:play-services-vision:17.0.2.
      > Could not get resource 'https://jcenter.bintray.com/com/google/android/gms/play-services-vision/17.0.2/play-services-vision-17.0.2.pom'.
         > Could not HEAD 'https://jcenter.bintray.com/com/google/android/gms/play-services-vision/17.0.2/play-services-vision-17.0.2.pom'.
            > sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
   > Could not resolve com.google.android.gms:play-services-vision:17.0.2.
      > Could not get resource 'https://maven.google.com/com/google/android/gms/play-services-vision/17.0.2/play-services-vision-17.0.2.pom'.
         > Could not HEAD 'https://maven.google.com/com/google/android/gms/play-services-vision/17.0.2/play-services-vision-17.0.2.pom'.
            > sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
   > Could not resolve com.google.android.gms:play-services-vision:17.0.2.
      > Could not get resource 'https://jitpack.io/com/google/android/gms/play-services-vision/17.0.2/play-services-vision-17.0.2.pom'.
         > Could not HEAD 'https://jitpack.io/com/google/android/gms/play-services-vision/17.0.2/play-services-vision-17.0.2.pom'.
            > sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 20s
23 actionable tasks: 23 executed
```

**Expected Behaviour**

compile successfully

**Steps to Reproduce**

1. `npm install --save react-native-camera`
2. insert ` <uses-permission android:name="android.permission.CAMERA" />`  to `AndroidManifest.xml ` 
3. insert `missingDimensionStrategy 'react-native-camera', 'general'` to `android/build.gradle`
4. `react-native link react-native-camera`
5. `react-native run-android`

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

Ryon-Lee, 2019-04-18:
same issue x2

DerekFei, 2019-07-19:
same issue

DedoJr, 2019-07-26:
same issue

dlombardi, 2019-08-13:
bump for continuing issue

tejase, 2019-09-21:
same issue

Zainabalhaidary, 2019-09-24:
My issue turned out to be just network proxy. Once switched to open network it got resolved.

Obi1Kennoby, 2019-10-08:
It seems the com.google.android.gms:play-services-vision:17.0.2 isn't available any more.
If you look inside on build.gradle of react-native-camera module, you will find method
```
def safeExtGet(prop, fallback) {
    rootProject.ext.has(prop) ? rootProject.ext.get(prop) : fallback
}
```
and it's call
```
def googlePlayServicesVisionVersion = safeExtGet('googlePlayServicesVisionVersion', safeExtGet('googlePlayServicesVersion', '17.0.2'))
```
So, you need to override variable `googlePlayServicesVisionVersion` in your android/build.gradle

add the next line to your ext block
`googlePlayServicesVisionVersion = "18.0.0"`
or
`googlePlayServicesVisionVersion = "+"`

closed 2019-12-14 by stale[bot]