react-native-camera/react-native-camera #2154: More than one file was found with OS independent path 'META-INF/proguard/androidx-annotations.pro'

brayanL, 2019-03-19:
# Bug Report
**To Do First**
- [Yes] Did you try latest release?
- [No] Did you try master?
- [Yes] Did you look for existing matching issues?

**Platforms**
* Android

**Libraries version used**
* react-native-cli: 2.0.1
* react-native: 0.58.4
* react-native-camera: 2.0.1

**Description/Current Behaviour**
Thanks for you wonderfull work with this library, i get started working with this library, I did achieve to do the configurations for ios project, all is well, but on Android project breaking my mind, the project not works, raise this error.

```
Task :app:transformResourcesWithMergeJavaResForDebug FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:transformResourcesWithMergeJavaResForDebug'.
> More than one file was found with OS independent path 'META-INF/proguard/androidx-annotations.pro'
```

I have configured firebase for react-native-google-signin library, and everything was fine until I installed react-native-camera library.

I read about: https://github.com/invertase/react-native-firebase/issues/1155#issuecomment-425448901, but that solution throw another errores, maybe do you have any solutions when working with severals google services libraries?
 
Here is `android/build.gradle` file:
```
buildscript {
    ext {
        buildToolsVersion = "28.0.3"
        minSdkVersion = 16
        compileSdkVersion = 28
        targetSdkVersion = 27
        supportLibVersion = "28.0.0"
        googlePlayServicesVersion = "15.0.2"
        googlePlayServicesVisionVersion = "15.0.2"
    }
    repositories {
        google()
        jcenter()
    }
    dependencies {
        classpath 'com.android.tools.build:gradle:3.2.1'
        classpath 'com.google.gms:google-services:4.2.0'
        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
    }
}

allprojects {
    repositories {
        mavenLocal()
        google()
        jcenter()
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url "$rootDir/../node_modules/react-native/android"
        }
        maven { url "https://jitpack.io" }
        maven { url "https://maven.google.com" }
    }
}

task wrapper(type: Wrapper) {
    gradleVersion = '4.7'
    distributionUrl = distributionUrl.replace("bin", "all")
}

```
Here is `app/build.gradle`
```
apply plugin: "com.android.application"

import com.android.build.OutputFile

project.ext.react = [
    entryFile: "index.js"
]

apply from: "../../node_modules/react-native/react.gradle"

def enableSeparateBuildPerCPUArchitecture = false

/**
 * Run Proguard to shrink the Java bytecode in release builds.
 */
def enableProguardInReleaseBuilds = false

android {
    compileSdkVersion rootProject.ext.compileSdkVersion
    buildToolsVersion rootProject.ext.buildToolsVersion

    defaultConfig {
        applicationId "com.buyers_app"
        minSdkVersion rootProject.ext.minSdkVersion
        targetSdkVersion rootProject.ext.targetSdkVersion
        missingDimensionStrategy 'react-native-camera', 'mlkit'
        versionCode 1
        versionName "1.0"
    }
    signingConfigs {
       release {
         storeFile file(buyers_app_RELEASE_STORE_FILE)
         storePassword System.getenv(buyers_app_RELEASE_STORE_PASSWORD)
         keyAlias buyers_app_RELEASE_KEY_ALIAS
         keyPassword System.getenv(buyers_app_RELEASE_KEY_PASSWORD)
      }
    }
    splits {
        abi {
            reset()
            enable enableSeparateBuildPerCPUArchitecture
            universalApk false  // If true, also generate a universal APK
            include "armeabi-v7a", "x86", "arm64-v8a"
        }
    }
    buildTypes {
        release {
            minifyEnabled enableProguardInReleaseBuilds
            proguardFiles getDefaultProguardFile("proguard-android.txt"), "proguard-rules.pro"
            signingConfig signingConfigs.release
        }
        profile {
            // Specifies a sorted list of fallback build types that the
            // plugin should try to use when a dependency does not include a
            // "staging" build type. You may specify as many fallbacks as you
            // like, and the plugin selects the first build type that's
            // available in the dependency.
            matchingFallbacks = ['debug', 'release']
        }
    }
    // applicationVariants are e.g. debug, release
    applicationVariants.all { variant ->
        variant.outputs.each { output ->
            // For each separate APK per architecture, set a unique version code as described here:
            // http://tools.android.com/tech-docs/new-build-system/user-guide/apk-splits
            def versionCodes = ["armeabi-v7a":1, "x86":2, "arm64-v8a": 3]
            def abi = output.getFilter(OutputFile.ABI)
            if (abi != null) {  // null for the universal-debug, universal-release variants
                output.versionCodeOverride =
                        versionCodes.get(abi) * 1048576 + defaultConfig.versionCode
            }
        }
    }
}

dependencies {
    implementation project(':react-native-camera')
    implementation project(':react-native-vector-icons')
    implementation project(':react-native-google-signin')
    implementation project(':react-native-gesture-handler')
    implementation fileTree(dir: "libs", include: ["*.jar"])
    implementation "com.android.support:appcompat-v7:${rootProject.ext.supportLibVersion}"
    implementation "com.facebook.react:react-native:+"  // From node_modules
    implementation "com.google.firebase:firebase-core:16.0.8"
}

// Run this once to be able to run the application with BUCK
// puts all compile dependencies into folder libs for BUCK to use
task copyDownloadableDepsToLibs(type: Copy) {
    from configurations.compile
    into 'libs'
}

configurations.all {
    resolutionStrategy {
        force 'com.android.support:support-v4:28.0.0'
    }
}

apply plugin: 'com.google.gms.google-services'
```
**Note:**
if remove configurations.all on app/build.gradle, that throw another errores:
```
* What went wrong:
Execution failed for task ':app:preDebugBuild'.
> Android dependency 'com.android.support:support-v4' has different version for the compile (26.1.0) and runtime (28.0.0) classpath. You should manually set the same version via DependencyResolution
```

dov11, 2019-03-19:
To fix your original error try adding the following to your `app/build.gradle`
```
    packagingOptions {
        exclude 'META-INF/proguard/androidx-annotations.pro'
    }
```
the above goes inside android block, like this:
```
android {
    ...
    packagingOptions {
        exclude 'META-INF/proguard/androidx-annotations.pro'
    }
}
```

See this SO answer: https://stackoverflow.com/a/52523819

brayanL, 2019-03-19:
You are a hero, thanks bro..!

closed 2019-03-20 by sibelius

farrukhhashmi65, 2019-04-06:
Still facing the issue,.. stuck to it
More than one file was found with OS independent path 'META-INF/androidx.exifinterface_exifinterface.version'

Thomazella, 2019-04-07:
I'd say try 
`exclude 'META-INF/androidx.exifinterface_exifinterface.version`

ha-younes72, 2019-04-10:
I still have stuck with this problem

dov11, 2019-04-14:
@farrukhhashmi65 , @ha-younes72 
Please post your stacktrace, and `app/build.gradle`.
You might want to add multiple such lines, e.g.:
```
packagingOptions {
        exclude 'META-INF/proguard/androidx-annotations.pro'
        exclude 'META-INF/androidx.exifinterface_exifinterface.version
}
```
If build is still failing, carefully read the error description - it might look the same because it still contains `META-INF`, but actually be a different one - keep adding the lines of this type.

issamabdelkrim, 2019-10-18:
add this to app/build.gradle 
packagingOptions {
       pickFirst 'lib/x86/libswscale.so'
       pickFirst 'lib/x86/libavcodec.so'
       pickFirst 'lib/x86/libavformat.so'
       pickFirst 'lib/x86/libavutil.so'
       pickFirst 'lib/x86_64/libavutil.so'
       pickFirst 'lib/x86_64/libavcodec.so'
       pickFirst 'lib/x86_64/libswscale.so'
       pickFirst 'lib/x86_64/libavformat.so'
       pickFirst 'lib/arm64-v8a/libavcodec.so'
       pickFirst 'lib/arm64-v8a/libavformat.so'
       pickFirst 'lib/arm64-v8a/libavutil.so'
       pickFirst 'lib/arm64-v8a/libswscale.so'
       pickFirst 'lib/armeabi-v7a/libavutil.so'
       pickFirst 'lib/armeabi-v7a/libavformat.so'
       pickFirst 'lib/armeabi-v7a/libavcodec.so'
       pickFirst 'lib/armeabi-v7a/libswscale.so'

   }

giovannitubera, 2020-03-20:
Hi, where can I find the app/build.gradle? I'm new to this and I having the same issue 

* What went wrong:
Execution failed for task ':app:transformResourcesWithMergeJavaResForDebug'.
> More than one file was found with OS independent path 'META-INF/proguard/androidx-annotations.pro'

M-Abd, 2021-04-04:
> To fix your original error try adding the following to your `app/build.gradle`
>
> ```
>     packagingOptions {
>         exclude 'META-INF/proguard/androidx-annotations.pro'
>     }
> ```
>
> the above goes [...]

this solution doesn't work in my case. i don't know why.
can you help me? please

ayezarniaung, 2022-03-31:
Update your Kotlin version and Gradle version in build.gradle .

EG.

ext.kotlin_version = '1.6.10'<= latest version

I solved this issues this way.