react-native-camera/react-native-camera #2155: Error while updating property 'googleVisionBarcodeType' of a view - BarcodeScanner Android

otaviogaiao, 2019-03-19:
# Bug Report

- [x] Did you try latest release?
- [ ] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

**Versions**

- Android: Android 6, Android 7
- react-native-camera: 2.0.1
- react-native: ExpoKit 32 (React Native 0.57.2)
- react: 16.5.0

**Description/Current Behaviour**
When I try to use the barcode scanner, the app crashes. It happens as soon as the screen mounts. If I remove the 'onGoogleVisionBarcodesDetected' and 'googleVisionBarcodeType'. The camera works (haven't tried to take pics since its not my intent). Methods pausePreview and resumePreview working fine.

The error I get is the following:

[image]

**Expected Behaviour**
Screen containing a BarcodeScanner should mount.

**Steps to Reproduce**
Rendering a RNCamera component with onGoogleVisionBarcodesDetected and googleVisionBarcodeType.

Here is a piece of code:

```  render() {
    return (
      <View style={styles.container}>
        <RNCamera
          style={styles.preview}
          type={RNCamera.Constants.Type.back}
          ref={this.cameraRef}
          captureAudio={false}
          {...Platform.select({
            android: {
              onGoogleVisionBarcodesDetected: this.onCodeScanned,
              googleVisionBarcodeType: [
                RNCamera.Constants.GoogleVisionBarcodeDetection.BarcodeType.QR_CODE,
              ],
            },
            ios: {
              onBarCodeRead: this.onCodeScanned,
              barCodeTypes: [RNCamera.Constants.BarCodeType.qr],
            },
          })}
        >
          <NavigationEvents onWillFocus={this.enableCam} onWillBlur={this.disableCam} />
          <TouchableOpacity style={styles.header} onPress={this.goBack} hitSlop={hitSlop15}>
            <MaterialCommunityIcons name="arrow-left" color={colors.branco} size={wp('7.4%')} />
            <Text style={styles.textVoltar}>Voltar</Text>
          </TouchableOpacity>
          {this.renderBody()}
        </RNCamera>
      </View>
    );
  }
```

Expo Camera has a totally different problem, which forced me to eject and try react-native-camera.

dov11, 2019-03-20:
@otaviogaiao, currently googleVisionBarcodeType only supports one constant to be passed, not an array, e.g. 
```
<RNCamera googleVisionBarcodeType={RNCamera.Constants.GoogleVisionBarcodeDetection.BarcodeType.QR_CODE} />
```
If you need multiple types, currently you will need to omit `googleVisionBarcodeType` prop altogether - this will set the detector to recognize all formats. 

I might make a PR to allow for an array of constants later.

otaviogaiao, 2019-03-20:
> @otaviogaiao, currently googleVisionBarcodeType only supports one constant to be passed, not an array, e.g.
>
> ```
> <RNCamera [...]

If I use a constant, I get another error (the same one I get if I use onBarCodeRead instead GoogleVision):

[image]

Any solution? Any help is appreciated. This is all I need to launch my app 😞

Thank you

sibelius, 2019-03-20:
@otaviogaiao check this https://github.com/react-native-community/react-native-camera/pull/2147

otaviogaiao, 2019-03-20:
> @otaviogaiao check this #2147

Thanks. I will point my project to that branch and leave the results here

otaviogaiao, 2019-03-20:
> @otaviogaiao check this #2147

Testing with that branch I get:

java.lang.NoSuchMethodError: No virtual method getCameraOrientation()I in class     Lorg/reactnative/camera/RNCameraView; or its super classes (declaration of 'org.reactnative.camera.RNCameraView' appears in /data/app/br.com.baladapp.produtor-1/split_lib_slice_4_apk.apk)
        at org.reactnative.camera.RNCameraView$1.onFramePreview(RNCameraView.java:125)
        at com.google.android.cameraview.CameraView$CallbackBridge.onFramePreview(CameraView.java:604)
        at com.google.android.cameraview.Camera1.onPreviewFrame(Camera1.java:766)
        at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1110)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:154)
        at android.app.ActivityThread.main(ActivityThread.java:6077)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:866)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:756)

sibelius, 2019-03-20:
this does not sounds related to that pull requests, do a clean install 

does this happen on master as well?

otaviogaiao, 2019-03-20:
> this does not sounds related to that pull requests, do a clean install
> 
> does this happen on master as well?

Just did a clean and tested from master and from the pull request again.

Issue happens on both.

java.lang.NoSuchMethodError: No virtual method getCameraOrientation()I in class Lorg/reactnative/camera/RNCameraView; or its super classes (declaration of 'org.reactnative.camera.RNCameraView' appears in /data/app/br.com.baladapp.produtor-1/split_lib_slice_4_apk.apk)
at org.reactnative.camera.RNCameraView$1.onFramePreview(RNCameraView.java:125)
at com.google.android.cameraview.CameraView$CallbackBridge.onFramePreview(CameraView.java:604)
at com.google.android.cameraview.Camera1.onPreviewFrame(Camera1.java:766)
at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1110)
at android.os.Handler.dispatchMessage(Handler.java:102)
at android.os.Looper.loop(Looper.java:154)
at android.app.ActivityThread.main(ActivityThread.java:6077)
at java.lang.reflect.Method.invoke(Native Method)
at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:866)
at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:756)

Do you think it may related to conflicts and bad dependency resolution with griddle? I don't know much about native android so I'm guessing.

sibelius, 2019-03-20:
it looks like this is another issue

the method is there https://github.com/react-native-community/react-native-camera/search?q=getCameraOrientation&unscoped_q=getCameraOrientation

otaviogaiao, 2019-03-20:
> it looks like this is another issue
> 
> the method is there https://github.com/react-native-community/react-native-camera/search?q=getCameraOrientation&unscoped_q=getCameraOrientation

I see. I will manually link it again and see if it resolves. In the meanwhile, do you have any suggestion ?

Thank you very much by your help by the way.

closed 2019-06-07 by stale[bot]

vetrivendhan48, 2019-12-06:
Guys, Is this issue fixed?