react-native-camera/react-native-camera #2156: The camera view freezes when switching back to the containing react native app

pramodxyle, 2019-03-20:
# Bug Report
**To Do First**
- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Related Modules**

- RNCamera

**Platforms**

- iOS

**Versions**

- iOS:
- react-native-camera:

**Description/Current Behaviour**

- The camera view freezes when switching back to the containing react native app.

**Expected Behaviour**

- The camera view should open with a preview.

**Steps to Reproduce**

1. Just use the example code here (https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md) in an expo app.

2. Minimize the app on iOS or switch to some other app.

3. Open the camera app. The camera preview is frozen.

pramodxyle, 2019-03-20:
For some reason, if I flip the camera the view starts working properly. If I now switch to another app and come back again, I resume the camera preview.
How can I handle this ?

pramodxyle, 2019-03-20:
Is this with the mount and unmount ? Looks like Camera from Expo handles this properly. But I don't want to use Camera from Expo. Want to use RNCamera from react-native-camera instead.

n1ru4l, 2019-03-20:
:tada: This issue has been resolved in version 2.0.2 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera)
- [GitHub release](https://github.com/react-native-community/react-native-camera/releases/tag/v2.0.2)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket: