react-native-camera/react-native-camera #2161: iOS doesn't show '⏎' in text detection response but Android does

tushgup, 2019-03-22:
# Bug Report
**To Do First**
- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Related Modules**
RNCamera
TextDetector

**Platforms**
iOS

**Versions**
- iOS: 12.1.4
- react-native-camera: 2.0.2
- react-native: 0.58.4
- react: 16.6.3

**Description/Current Behaviour**
I'm building a text detection app using RNCamera. I switched to v2 recently because of the MLKit implementation. While studying the text response from the `textBlocks` object, I noticed that the iOS response isn't returning the '⏎' / enters in between the text but the Android response is. 

Here are the console logs:

Here's what I'm pointing my camera towards:
[image]

iOS:
[image]

Android:
[image]

**Expected Behaviour**
The iOS response should also return the enters/next line identifier or the '⏎' in between the console.logs like Android does.

**Steps to Reproduce**
Try scanning this piece of text on Android and iOS and log the console response.
[image]

tushgup, 2019-03-22:
@dov11 I think you'd be the best person to have a look at this since you majorly implemented the v2 MLKit rewrite of Text Detection.

dov11, 2019-03-24:
Hi @tushgup, thanks for reporting this interesting issue. 

Looks like text property of FIRVisionTextBlock does not contain newline characters, even though 3 separate lines were detected.
I have checked official native [Firebase example](https://github.com/firebase/quickstart-ios/tree/master/mlvision) for iOS, and the newline characters are absent there as well, so looks like it is not our implementation that is a problem, but MLKit differences for different platforms. (i.e. `FirebaseVisionText.TextBlock.getText()` vs `FIRVisionTextBlock.text`)

If you need to detect where the text has a new line, I would suggest using line components of the text block. In your screenshot from iOS, you can see `components` array, which contains 3 lines. If you want the same result as the one on android, you can use something like 
```const value = textBlock.components.map(component => component.value).join('⏎')```
instead of relying on `textBlock.value`.

Hope that helps.

closed 2019-06-07 by stale[bot]