react-native-camera/react-native-camera #2164: When the camera type is changed, video recording stops.

furkancelik, 2019-03-25:
# Bug Report
**To Do First**
- [X] Did you try latest release?
- [X] Did you try master?
- [X] Did you look for existing matching issues?

**Related Modules**
  RNCamera

**Platforms**

Android
iOS

**Versions**

- Android:
API Levels: 23, 25, 26, 27, 28
Build Tools: 23.0.1, 25.0.0, 26.0.1, 26.0.2, 26.0.3, 27.0.3, 28.0.3
System Images: android-27 | Intel x86 Atom_64, android-27 | Google APIs Intel x86 Atom, android-28 | Google APIs Intel x86 Atom
- iOS:
Platforms: iOS 12.1, macOS 10.14, tvOS 12.1, watchOS 5.1
- react-native-camera:
^2.0.2
- react-native:
0.59.0 => 0.59.0 
- react:
 16.8.3 => 16.8.3 

**Description/Current Behaviour**
When I change the camera type (front->rear / rear -> front) when recording a video, the video recording stops.
"Video is not recording "error.

I think the error is on this line;
https://github.com/react-native-community/react-native-camera/blob/master/ios/RN/RNCamera.m#L149

After RNCamera was rendered, it seemed to be rendered again when the type changed. when the camera type is changed, the reference value is reassigned.
this and related lines need to be corrected...

**Expected Behaviour**
When I change the camera type when recording a video, it should continue recording the video.
we can switch to the front camera when recording video in the Instagram Stories.

Sarvarr, 2019-08-09:
same issue

Sarvarr, 2019-08-09:
any solution?

BaderSerhan, 2019-11-02:
Why was this closed? This is a genuine issue...

n-popov-geniusee, 2020-07-07:
the same issue for us too

shrikanttapkeer07, 2020-08-05:
ANy update on this having the same issue ??

closed 2020-10-11 by stale[bot]

jaroslav009, 2021-03-31:
same issue

Nahani, 2021-06-27:
same issue