react-native-camera/react-native-camera #2165: Avoid including mlkit face model

mllrmat, 2019-03-25:
# Question

**Ask your Question**
How can I avoid that the mlkit face detection model is included in the APK? I am just using barcode scanning capabilities, and the face model grows my APK by 9MB. 

**Tags**
Android, mlkit, apk size

sibelius, 2019-03-25:
try this

https://github.com/react-native-community/react-native-camera/blob/master/examples/basic/android/app/build.gradle#L106

`missingDimensionStrategy 'react-native-camera', 'general'`

this will avoid adding `MLKit` to your android app

mllrmat, 2019-03-25:
Ok. I thought that mlkit will greatly improve the barcode detection capabilities. Shouldn't the face detection model only be added if face detection is enabled? 

eg. This line in the modules build.gradle removed in all other cases: 
mlkitImplementation "com.google.firebase:firebase-ml-vision-face-model:17.0.2"

sibelius, 2019-03-25:
I don't think we can detect this on build time

user needs to explicit opt in or opt out of this feature

mllrmat, 2019-03-25:
Couldn't you add a gradle buildType/flavor similar to the ios subspecs?

sibelius, 2019-03-25:
isn't `missingDimensionStrategy` the same?

mllrmat, 2019-03-25:
As far as I understand it this way you opt-out of mlkit completely, what I am asking for is to use mlkit for barcode recognition without importing the face model (I do not use face recognition). This way I can use the upside of mlkit without increasing the APK size dramatically. 
One possible solution could be to import react-native-camera with exclude of the face model: 
```
implementation (project(':react-native-camera')){
    exclude group: 'com.google.firebase', module: 'firebase-ml-vision-face-model'
}
```

sibelius, 2019-03-25:
feel free to send a pull request if you find a nice solution for your problem

mllrmat, 2019-03-25:
For me the above works. But I guess the better solution would be to have a productFlavor for mlkit text + barcode and another for face recognition. But I am not an gradle expert.

mllrmat, 2019-03-25:
Thank you for your quick support!

closed 2019-03-25 by mllrmat