react-native-camera/react-native-camera #2170: What steps are required to read qr-code on iOS?

Bardiamist, 2019-03-26:
# Question

**To Do First**
- [x] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [x] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [x] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**
Can I read qr-code on iOS using **react-native-camera 2.x** without CocoaPods?
What steps from readme required for read qr-code?

[I created feedback](https://github.com/react-native-community/react-native-camera/issues/2019#issuecomment-475293719) which described in detail why it is not clear.

**Tags**
pods, qr-code, ios

Bardimaswift, 2019-03-28:
@dov11 Can you answer?

dov11, 2019-03-28:
Hi @Bardimaswift, currently on iOS barcode detection using googleVision is not yet implemented, you have to use `onBarCodeRead`. You don't need any special installation procedure, and it should work out of the box, same as on version 1.x. I just tested this by building basic example, had to change `onGoogleVisionBarcodesDetected` to `onBarCodeRead` prop, and use `barcodeRecognized = (barcodes) => console.log(barcodes)`, because return value is not an array of barcodes as in Google vision case, but just one object. Apart from that it seems to be working ok, without the need for Cocoapods/firebase setup.

I do agree with you that we should unify barcode scanning behaviour on both platforms, and it will likely happen when MLKit barcode will be implemented on iOS - onBarCode returns a bit too different results for iOS and Android (but I haven't looked much into its implementation, to be fair). After the implementation docs can be cleaned up.

Regarding your feedback: 
>Text recognition is currently available only via CocoaPods Path. It bad. I do not want to use Pods

Seems like CocoaPods will be the way to go, from React native blog: [We plan on recommending CocoaPods by default for iOS projects](https://facebook.github.io/react-native/blog/2019/03/01/react-native-open-source-update)
And I think adding and managing optional modules is much easier this way, (vs postinstall scipts, for example)

>It is not clear how friendly react-native-camera with react-native-firebase

You don't need react-native-firebase to use react-native-camera (as demonstrated in mlkit example here), and if you do need both packages, they work fine alongside each other (tested in 2 apps in production).

closed 2019-04-02 by Bardiamist