react-native-camera/react-native-camera #2171: Flash not working with Video

cristianocca, 2019-03-27:
# Bug Report

**Related Modules**

RNCamera

**Platforms**
Android

**Versions**
- Android: 7
- react-native-camera: 2.0.2
- react-native: 0.59.2
- react: 16.8.5

Using `<RNCamera flashMode={'on'} ...others/>` and even loading flashMode from a state variable that is changed on demand, the flash is always off when recording a video with `camera.recordAsync({quality: '480p'})`

Funny thing is, even if flash is set to torch, it gets immediately turned off when recording starts, and turned on again when recording stops. Should a flash option be set when calling recordAsync?

cristianocca, 2019-05-10:
Any updates on this? Is it a library restriction instead?

closed 2019-07-16 by stale[bot]

cristianoccazinsp, 2019-10-16:
https://github.com/react-native-community/react-native-camera/pull/2543 fixes torch while recording on both Android and iOS