react-native-camera/react-native-camera #2173: takePicture or capture too delay and Picture resolution always set to 16:9

rey501, 2019-03-27:
**Which implementation are you using**
RNCamera OR RTCamera

**Steps to reproduce**
Take a picture with android 

```
takePicture=() =>{
    const options = {jpegQuality:100};
    this.camera.capture(options)    
    .then(data => {
      console.log(data);
     
      this.startscanmethod(data.path);
    })
    .catch(err => alert(err));

}
```

**Expected behavior**

1. The picture capture speed must be same as device default camera.
2. picture taken by RNCamera(Camera) preview should have same resolution(16:9, 4:3 , etc) as device camera or has option to set by prophetically.

**Actual behaviour**

1. The picture taken process to much delay almost 6-8 second.
2. Camera preview always preview in picture with 16:9 ratio.

**Environment**

- Redmi 4a  Android-7.1.1 | VIVO Andorid-7.1.1
- react-native-camera : ^1.1.4
- react: 16.6.3
- react-native: 0.57.8

I have added the option ratio(4:3) as well, which didn't make any difference

closed 2019-06-07 by stale[bot]

devilabhi, 2019-10-07:
Capturing of picture is taking 6-8 seconds. Did you find any solution?

lynmije, 2019-12-06:
> Capturing of picture is taking 6-8 seconds. Did you find any solution?

Have you already any solutions with these issues? I have also encountered this.