react-native-camera/react-native-camera #2175: Android Studio new version error

Niroj11, 2019-03-28:
I have encountered this error after I upgraded to new version of android studio

Unable to resolve dependency for ':app@debug/compileClasspath': Could not resolve project :react-native-camera.
Open File
Show Details

Gradle not able to sync

sibelius, 2019-03-28:
Post full stack trace

lessp, 2019-04-02:
Running in to the same error with React Native 0.59, any ideas?

Niroj11, 2019-04-02:
I setup a fresh react native project with all the latest version of the packages that I was using. It still showed me some errors in version 0.59 so I downgraded to 0.58 and it is working fine now. I could not solve the issue so I copied all my project into a new fresh project

closed 2019-06-08 by stale[bot]