react-native-camera/react-native-camera #2176: mirrorVideo doens't work for Videos - rn 0.59.1 & Android 9

emilemoureau, 2019-03-29:
**- startRecording**

```javascript
this.camera.recordAsync({ mirrorVideo: true });
```

"react-native-video": "^4.4.0",

**- renderVideo**

```javascript
<Video 
  source={{ uri: this.state.uri}}
  ref={(ref) => {
   this.player = ref
   }}
  fullscreen={false}
  repeat={true}
  volume={1.0}
/>
```

Everything works perfectly on iOS!

susanapons89, 2019-04-18:
Having the same issue. Did you find a solution?

emilemoureau, 2019-04-20:
Not really..

Try to work around the problem with "transform" :

> https://facebook.github.io/react-native/docs/transforms

```javascript
<Video
  source={{ uri: uri }}
  style={{
    transform: [{ rotate: '270deg'}, { scaleX: -1.8 }, { scaleY: 1.8}]
  }}
/>
```

victorwpbastos, 2019-06-22:
same problem here.

asarasua, 2019-07-30:
Same issue here, adding `mirrorVideo: true` in the options has no effect.

closed 2019-10-05 by stale[bot]