react-native-camera/react-native-camera #2177: example not working - android

ggunti, 2019-03-31:
I just tried to run the mlkit example for android following the documentation: https://github.com/react-native-community/react-native-camera/tree/master/examples/mlkit

But this is what I get:
[image]

sibelius, 2019-03-31:
Have you ran yarn on both root and examples projects?

ggunti, 2019-03-31:
@sibelius you are right. You solved the problem. But now just appeared a new one:
Inside `./mlkit/android/app/build.gradle` I just changed the applicationId from "com.mlkit" into "com.gothasoft.react_native_camera_example" which is my app id provided by firebase - seems that it is required to change it (see screenshot 1)

Screenshot 1:
[image]

Now, the problem is the following:
[image]

It says that MainActivity does not exist.

dov11, 2019-04-05:
You need to change `package com.mlkit;` to `package com.gothasoft.react_native_camera_example;`
In MainActivity.java and MainApplication.java. Change the folder structure of these files to
```
com
|
 __gothasoft
  |
   __react_native_camera_example
```
You will need to rename package in your android manifest 
```
<manifest xmlns:android="http://schemas.android.com/apk/res/android"
  package="com.mlkit">
```

closed 2019-06-11 by stale[bot]