react-native-camera/react-native-camera #2178: Pinch to zoom with RN Camera

rsimbu89, 2019-03-31:
# Question
Hi Team,
 Is there any way to integrate Pinch to zoom the camera view for both iOS and Android with RNCamera?

sibelius, 2019-03-31:
There are some snippet to do this, check issues

rsimbu89, 2019-04-01:
@sibelius  
I tried this ZoomView implementation https://github.com/react-native-community/react-native-camera/issues/1282, But not satisfies with the zoom result. Is
 there any better solution than this?

rsimbu89, 2019-04-07:
Team,
 Is there any other better solutions?

FaggioniHQ, 2019-06-12:
I implemented this functionality using react-native-gesture-handler module. But is not good enough, the camera does not respond well under continues changes on the zoom prop. Is there any option for this feature? Thanks in advanced

Kuhne1, 2019-06-20:
@FaggioniHQ Would you be able to share your implementation using the gesture handler? 
Thanks :)

FaggioniHQ, 2019-06-27:
Sorry for the late reply,

Here is the code.

```js
import React, {Component} from "react";
import {PinchGestureHandler, State} from "react-native-gesture-handler";

class PinchGesture extends Component{
    constructor(props){
        super(props);
    }
    onPinchGestureEvent({nativeEvent}){
        if (nativeEvent.state === State.ACTIVE) {
            this.props.onPinchScreen(nativeEvent);
        }
    }
    render(){
        return(
            <PinchGestureHandler
                onGestureEvent={this.onPinchGestureEvent.bind(this)}>
                {this.props.children}
            </PinchGestureHandler>
            );
    }
}
export default PinchGesture;
```

```js
import React, { Component } from "react";
import { View } from "react-native";
import { cameraStyles as styles } from './styles';
import BackButtonCamera from '../buttons/BackButtonCamera';
import { RNCamera } from 'react-native-camera';
/*
 * Buttons
 */
import TakeAPicture from '../../components/buttons/camera/TakeAPicture';
import FlashButton from '../../components/buttons/camera/FlashButton';
import ChangeCamera from'../../components/buttons/camera/ChangeCamera';
import GalleryButton from '../../components/buttons/camera/GalleryButton';
import CameraZoomSlider from '../../components/sliders/CameraZoomSlider';
/**
 * Gestures
 */
import LongTapGesture from '../../components/gestures/LongTapGesture';
import PinchGesture from '../../components/gestures/PinchGesture';

/**
 * Handler
 */
import CameraHandler from '../../modules/helpers/periphericals/CameraHandler';
import RecordBlinkIndicator from "../blinkers/RecordBlinkIndicator";
import VibrationHandler from '../../modules/helpers/periphericals/VibrationHandler';
import SoundHandler from '../../modules/helpers/periphericals/SoundHandler';

class Camera extends Component {
    constructor(props) {
        super(props);
        this.styles = styles;
        this.zoomRate = 0.005;
        this.helper = new CameraHandler(this);
        this.vibrato = new VibrationHandler();
        this.sound = null;
        this.state = {
            // auto - on -off
            flashMode: RNCamera.Constants.FlashMode.auto,
            // front -> true , back -> false
            cameraMode: true,
[... 212 lines cut ...]
            this.camera.setState({
                flashButtonName: 'flash-on',
                flashMode: RNCamera.Constants.FlashMode.torch
            });
        }
    }
    resolvePinchGesture(nativeEvent){
        if(nativeEvent.scale > 1){
            if(this.camera.state.zoom + this.camera.zoomRate > 1){
                this.camera.setState({ zoom: 1 });
            }else{
                this.camera.setState({ zoom: this.camera.state.zoom + this.camera.zoomRate });
            }
        }else if(nativeEvent.scale < 1){
            if(this.camera.state.zoom - this.camera.zoomRate < 0){
                this.camera.setState({ zoom: 0 });
            }else{
                this.camera.setState({ zoom: this.camera.state.zoom - this.camera.zoomRate })
            }
        }
    }
}
export default CameraHandler;
```
Not sure if this the best channel for this. But how you guys deal with large files/videos on RN. How do you manage to upload those files in the background?. Thanks in advanced

MateusAndrade, 2019-07-01:
@FaggioniHQ you can give a look at [react-native-fetch-blob](https://github.com/joltup/rn-fetch-blob) and [react-native-fs](https://github.com/itinance/react-native-fs).

FaggioniHQ, 2019-07-01:
Yeah, I'm using react-native-fetch-blob already. But seems like this process its a bit slow... moving the videos/images to the filesystem and uploading the file to our backend. I implemented a file queue but seems to take a bit of time. Let me know if you guys have a better approach to this functionality.

Thanks in advanced

MateusAndrade, 2019-07-01:
It can be related to How react-native works. Btw, i recommend reading about that when working with process that demand a higher process. :smile:

d-dizhevsky, 2019-08-29:
@FaggioniHQ thanks for the code snippet. I managed to make it work better with continuous pitch events by changing `if(nativeEvent.scale > 1)` to `if (nativeEvent.scale > this.prevZoomScale)` and setting `this.prevZoomScale = nativeEvent.scale` at the bottom of the `resolvePinchGesture` in your case.
Also, keep in mind that zoom min/max is different on iOS and Android. See this comment for reference: https://github.com/react-native-community/react-native-camera/issues/1772#issuecomment-419068946

This is how I handle it (only for iOS atm):
```
    onPitchGestureEventHandler = ({ nativeEvent }) => {
        if (!nativeEvent || nativeEvent.state !== State.ACTIVE) {
            return;
        }

        if (nativeEvent.scale > this.prevZoomScale) {
            this.setState({ zoom: Math.min(0.05, this.state.zoom + this.zoomRate) });
        } else {
            this.setState({ zoom: Math.max(0, this.state.zoom - (this.zoomRate + this.zoomRate / 2)) });
        }

        this.prevZoomScale = nativeEvent.scale;
    }
```

FaggioniHQ, 2019-08-29:
Thanks for the feedback. I will check this later. 👍

fabriziobertoglio1987, 2019-10-17:
@FaggioniHQ @d-dizhevsky 

this is my existing code adapted and cleaned [from the following example](https://github.com/react-native-community/react-native-camera/issues/2178#issuecomment-506547158). Can you help me completing this functionality? Thanks a lot !

```
import React, { Component } from 'react';
import { View } from 'react-native';
import { PinchGestureHandler, State } from 'react-native-gesture-handler';

export default class ZoomView extends Component {
  onGesturePinch = ({ nativeEvent }) => {
    this.props.onPinchProgress(nativeEvent.scale)
  }

  onPinchHandlerStateChange = event => {
    const pinch_start = event.nativeEvent.state === State.END
    const pinch_begin = event.nativeEvent.oldState === State.BEGAN
    const pinch_active = event.nativeEvent.state === State.ACTIVE
    if (pinch_start) {
      this.props.onPinchEnd()
    }
    else if (pinch_begin && pinch_active) {
      this.props.onPinchStart()
    }
  }

  render() {
    const { style } = this.props
    return (
      <PinchGestureHandler
        onGestureEvent={this.onGesturePinch}
        onHandlerStateChange={this.onPinchHandlerStateChange}>
        { this.props.children }
      </PinchGestureHandler>
    )
  }
}
```

and the `Camera` Component

```
import React, { Component } from 'react'
import { RNCamera } from 'react-native-camera';
iimport ZoomView from './ZoomView';

export default class Recorder extends Component {
  _render() {
    return (
      <ZoomView
        onPinchStart={() => {}}
        onPinchEnd={() => {}}
        onPinchProgress={zoom => this.setState({ zoom })}>
        <RNCamera
          ref={ref => {
            this.camera = ref;
          }}
          zoom={this.state.zoom} />
      </ZoomView>
    )
  }
}
```

fabriziobertoglio1987, 2019-10-18:
Thanks @cristianoccazinsp for your [very complete and clear solution](https://github.com/react-native-community/react-native-camera/issues/1282#issuecomment-543188018). I look forward to contribute to this amazing project

fabriziobertoglio1987, 2019-10-18:
the solution was provided by @cristianoccazinsp with the following [answer](https://github.com/react-native-community/react-native-camera/issues/1282#issuecomment-543188018)

Initial testing without using `this.camera.setNativeProps({zoom: zoom})` have smooth zoom effect on even old android and iphone 5 devices.

Thanks a lot for the support!
[image]

sibelius, 2019-10-18:
can someone add this to the docs?

fabriziobertoglio1987, 2019-10-19:
credits go to @cristianoccazinsp with the [following answer](https://github.com/react-native-community/react-native-camera/issues/1282#issuecomment-543188018) and the ongoing discussion at [#1282](https://github.com/react-native-community/react-native-camera/issues/1282#issuecomment-543872120)

I still need to test the functionality with `setNativeProps` and I am not sure that the solution will work for all devices. I currently can only test it on iPhone 5 and Android Oppo A83.

iPhone 5 has relatively small screen so it not a good subject for testing the funcionality.

Thanks a lot

closed 2019-12-25 by stale[bot]

fabriziobertoglio1987, 2020-01-14:
@sibelius I am searching for a pull request assignment (writing code). Would you consider having this functionality in this library or as a separate npm package? Thanks a lot

sibelius, 2020-01-14:
@fabriziobertoglio1987 I think it is better as a new package, so we can reuse in expo camera as well

let me know if you need any help

cristianoccazinsp, 2020-01-14:
Considering the extra dependencies (gesture handler?), definitely looks like something that should be added as another library. Not sure if it would be worth it trying to implement it with react native's pan handlers instead.

sibelius, 2020-01-14:
we can link to the docs to make it easier to find it

fabriziobertoglio1987, 2020-01-14:
Thanks @sibelius @cristianocca I started my research. 
I currently believe that the only viable solution is making a pull request.
I started reading some documentation on how to implement this feature on android. I'm not an Android Expert and I can give up on this if that is your advice. 

As my understanding everything required for zooming is already present in the [`CameraViewManager` method `setZoom()`](https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/org/reactnative/camera/CameraViewManager.java#L116) available inside the `react-native-camera` npm package

The method [is exposed in react with `@ReactProp(name = "zoom")`](https://facebook.github.io/react-native/docs/native-components-android#3-expose-view-property-setters-using-reactprop-or-reactpropgroup-annotation)

```java
public class CameraViewManager extends ViewGroupManager<RNCameraView> {
  @ReactProp(name = "zoom")
  public void setZoom(RNCameraView view, float zoom) {
    view.setZoom(zoom);
  }
}
```

I believe that I should wrap a  [`ScaleGestureDetector` instance](https://developer.android.com/reference/android/view/ScaleGestureDetector) around the appropriate view and pass the correct value to the method `setZoom()`

Any suggestion is highly appreciated, I keep working on this until you tell me to stop.
I don't have any idea on how to implement this in a separate package and I am not planning to use ReactNativeGestureHandler.

Thanks a lot

phyzical, 2020-01-29:
@fabriziobertoglio1987 Hey is this contained in a package anywhere yet for testing (even if its rough)? if not any ETA if its sooonish i can hold off on writing my own implementation and just test what you come up with :)

Thanks!

fabriziobertoglio1987, 2020-01-29:
Thanks @sibelius and @phyzical , but after an [extensive research](https://github.com/react-native-community/react-native-camera/issues/2178#issuecomment-574228101), I don't believe that it makes sense creating a separate package for this functionality. The package would receive negative feedback from the community as it does not solve the **main problem**  [explained here](https://github.com/react-native-community/react-native-camera/issues/2178#issuecomment-574228101).

I agree with @sibelius that this functionality [should remain in user land](https://github.com/react-native-community/react-native-camera/issues/1282#issuecomment-392520205), but I would consider implementing `setNativeProps` as discussed [here](https://github.com/react-native-community/react-native-camera/issues/2178#issuecomment-574228101) (I may not be able to do this completely alone).

The **feature request is closed** and potential pull request will not be merged.

cristianoccazinsp, 2020-01-29:
One of the demo apps has a pinch to zoom implemented as well. You can
always copy paste ;)

SimonErm, 2020-04-30:
#2815