react-native-camera/react-native-camera #2179: Build Failed

shtengel, 2019-03-31:
Build is failing when adding react-native-camera as a dependency in build.gradle (app),
saw it happen to someone else here but no one replied... anyone know how to fix it ?
i tried to exclude com.facebook from react-native-camera but it does nothing 

# Bug Report
**To Do First**
- [ V] Did you try latest release?
- [ V] Did you try master?
- [V ] Did you look for existing matching issues?

**Related Modules**

**Platforms**

**Versions**
- Android: 28
- react-native-camera: 2.1.1
- react-native: 0.51.1
- react: 16.0.0

**Description/Current Behaviour**
AGPBI: {"kind":"error","text":"Program type already present: com.facebook.infer.annotation.Mutable","sources":[{}],"tool":"D8"}

Execution failed for task ':app:transformDexArchiveWithExternalLibsDexMergerForX86Debug'.
  Program type already present: com.facebook.infer.annotation.Mutable

**Steps to Reproduce**
Add react-native-camera as a dependency to gradle

shtengel, 2019-03-31:
commenting out `implementation 'com.facebook.infer.annotation:infer-annotation:+'` makes it to build succesfully

sibelius, 2019-03-31:
Where?

shtengel, 2019-03-31:
works if i change it to `implementation com.facebook.infer.annotation:infer-annotation:10.0.0`
in `react-native-camera->build.gradle`

Strate, 2019-04-01:
Can confirm that issue exists, and confirm that commenting out `implementation 'com.facebook.infer.annotation:infer-annotation:+'` fixes issue

sibelius, 2019-04-01:
can somebody send a pull request that solve this?

Strate, 2019-04-01:
@sibelius just with commenting line? I am not sure that exactly happens, for example I don't use mlkit, maybe this is related.

tadeumx1, 2019-04-01:
@sibelius I can work on this

shtengel, 2019-04-02:
If you comment out then you can build but then runtime exception is thrown ( in my case)

closed 2019-04-07 by sibelius

n1ru4l, 2019-04-07:
:tada: This issue has been resolved in version 2.2.2 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera)
- [GitHub release](https://github.com/react-native-community/react-native-camera/releases/tag/v2.2.2)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

PR #2194 (2019-04-07): replace assertion with a null check
fixes #2179
plus some small cleanup