react-native-camera/react-native-camera #2181: width parameter ignored if skipProcessing is true

cristianocca, 2019-04-01:
**Related Modules**
RNCamera

**Platforms**
Android

**Versions**

- Android: 7.x and 8.x
- react-native-camera: 2.2.0
- react-native: 0.59.3
- react: 16.8.6

Hello all,

If the "skipProcessing" flag is set to true, then the width parameter is ignored (i.e., the pictures are always taken at native resolution). 
I wonder if this is intentional or not. I would really want to be able to use skipProcessing: true since pictures are otherwise extremely slow to take.

mike-niemand, 2019-04-09:
This is the correct behaviour from my reading of the docs.

_skipProcessing (android only, boolean). This property skips all image processing on android, this makes taking photos super fast, but you loose some of the information, width, height and the ability to do some processing on the image (base64, width, quality, mirrorImage, exif, etc)_

cristianocca, 2019-04-09:
Not from my understanding. I would guess that the width and height information are not available after capture; however, setting the capture width and height should still be possible. 

Otherwise, does the plugin always take pictures in full resolution and then resize it if the width parameter is supplied? Any reason to not set the camera resolution before capturing?

sibelius, 2019-04-09:
Width is considered a processing operation

mike-niemand, 2019-04-09:
The problem for me is that I don't get width even if skipProcessing is false:

    const options = { quality: 0.8, skipProcessing: false, width: 1024 };
    await this.camera.takePictureAsync(options)
        .then(data => {
            console.log(data)
    });

Results in: `{uri: xxx, pictureOrientation: 1, deviceOrientation: 1}`

Surely it should return width?

mike-niemand, 2019-04-09:
> Not from my understanding. I would guess that the width and height information are not available after capture; however, setting the capture width and height should still be possible.
>
> Otherwise, [...]

May help: 

https://github.com/react-native-community/react-native-camera/issues/1271#issuecomment-367450669

Although, as per earlier post, I don't get width currently and it doesn't set width so I'm a bit stumped.

cristianocca, 2019-04-09:
So the only way to obtain a lower resolution photo in Android is to actually process it? It is amazing the time difference between skip processing and not. Makes the whole option to take a lower resolution photo (with width processing) quite useless since it takes 10 times more to process.

Are there any reasons that the photo quality can't be set on Android?

sibelius, 2019-04-09:
there is quality option

mike-niemand, 2019-04-09:
Is it a known issue that width isn't returned on skipProcessing: false?

Also, I've spotted a minor grammatical error in the docs. Must I submit a
pull or do you want to correct it, Sibelius? It's just 'loose' should be
'lose'.

😀

cristianocca, 2019-04-09:
> 
> 
> there is quality option

Correct me if I'm wrong, but this only changes the JPEG compression quality output, and the pictures will come still in full resolution.

mike-niemand, 2019-04-09:
> > there is quality option
> 
> Correct me if I'm wrong, but this only changes the JPEG compression quality output, and the pictures will come still in full resolution.

That is correct.  I've also tried:

    Image.getSize(data.uri, (width, height) => {
        console.log(width, height)
    })

But it's slow and yucky.

closed 2019-06-15 by stale[bot]