react-native-camera/react-native-camera #2184: Basic example doesn't work with XCode 10... (¿React Native upgrade issue?)

grisaitis, 2019-04-02:
# Bug Report
**To Do First**
- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?
- [ ] Did you sacrifice a bunny to the React Native gods?

**Related Modules**

**Platforms**

12.2, iPhone X simulator

**Versions**

- Android:
- iOS: 12.2
- react-native-camera: v2.2.0 (current master)
- react-native: 0.59.1
- react: 16.8.3

**Description/Current Behaviour**

I have encountered a few issues with the basic example:
1. The project doesn't build for iOS (`react-native run-ios` or in XCode) without first adding `JavaScriptCore.framework` to `Build Phases > Targets > RNCameraExample > Link Binary With Libraries`
2. the Metro Builder server fails with:
```
 BUNDLE  [ios, dev] ./index.js ░░░░░░░░░░░░░░░░ 0.0% (0/1)::1 - - [02/Apr/2019:19:49:30 +0000] "GET /index.bundle?platform=ios&dev=true&minify=false HTTP/1.1" 200 - "-" "RNCameraExample/1 CFNetwork/978.0.7 Darwin/18.2.0"
error: bundling failed: Error: Unable to resolve module `@babel/runtime/helpers/interopRequireDefault` from `/Users/william/src/react-native-camera.0/src/index.js`: Module `@babel/runtime/helpers/interopRequireDefault` does not exist in the Haste module map
```

And this is not fixed by any of the recommended remedies, like clearing watchman watches, etc. 

I rebooted my machine, created a fresh repo, etc and still encountered these issues. 

**Expected Behaviour**

**Steps to Reproduce**

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

grisaitis, 2019-04-02:
I can cure the metro bundler error above with `npm add @babel/runtime`. 

But then react itself isn't importable, or something:

```
Loading dependency graph, done.
::1 - - [02/Apr/2019:20:08:32 +0000] "GET /index.bundle?platform=ios&dev=true&minify=false HTTP/1.1" 200 - "-" "RNCameraExample/1 CFNetwork/978.0.7 Darwin/18.2.0"
Error: Unable to resolve module `./index` from `/Users/william/src/react-native-camera.0/examples/basic/node_modules/react-native/.`: The module `./index` could not be found from `/Users/william/src/react-native-camera.0/examples/basic/node_modules/react-native/.`. Indeed, none of these files exist:
  * `/Users/william/src/react-native-camera.0/examples/basic/node_modules/react-native/index(.native||.ios.js|.native.js|.js|.ios.json|.native.json|.json|.ios.ts|.native.ts|.ts|.ios.tsx|.native.tsx|.tsx)`
  * `/Users/william/src/react-native-camera.0/examples/basic/node_modules/react-native/index/index(.native||.ios.js|.native.js|.js|.ios.json|.native.json|.json|.ios.ts|.native.ts|.ts|.ios.tsx|.native.tsx|.tsx)`
    at ModuleResolver.resolveDependency (/Users/william/src/react-native-camera.0/examples/basic/node_modules/metro/src/node-haste/DependencyGraph/ModuleResolution.js:163:15)
........
```

grisaitis, 2019-04-03:
I have no problems if I simply create a fresh React Native app and copy and paste `examples/basic/App.js`.

h0rm, 2019-04-03:
which react-native version did you upgrade from? i fixed a similar issue by making the changed proposed by rn-diff-purge https://pvinis.github.io/purge-web/ (as recommended by react). JavaScriptCore.framework is not needed in earlier versions for instance

grisaitis, 2019-04-03:
No upgrade was involved. I was just trying to run the example from scratch. I was about to try the rn-diff approach as well, but solved the issue with a fresh `react-native init` before I got to that point. 

You're right about `metro.config.js`. A similar issue in the react-native repo (https://github.com/facebook/react-native/issues/23908) discusses that here: https://github.com/facebook/react-native/issues/23908#issuecomment-473204593

grisaitis, 2019-04-03:
I guess the example basically needs to be updated for the most recent react. 

What's the best way to update it? Migrate it with some tool like rn-diff-purge? Swap it out with a fresh project from `react-native init`?

h0rm, 2019-04-03:
according to the official bog https://facebook.github.io/react-native/blog/#upgrading-to-059 they improved the update experience with 0.59 and starting from 0.60 `react-native upgrade` (based on` rn-diff-purge`) can be used so these issues should be gone in the future. to upgrade to 0.59 the changes still have to be made manually using `rn-diff-purge` however...

closed 2019-06-14 by stale[bot]