react-native-camera/react-native-camera #2185: A problem occurred configuring project ':react-native-camera'.

xuanttph06854, 2019-04-03:
I install react-native-camera but I miss error:
A problem occurred configuring project ':react-native-camera'.
> Failed to notify project evaluation listener.
   > Android tasks have already been created.
     This happens when calling android.applicationVariants,
     android.libraryVariants or android.testVariants.
     Once these methods are called, it is not possible to
     continue configuring the model.
Please help me fix it. Tks

closed 2019-06-09 by stale[bot]

mg4u, 2019-06-28:
I have the same issue, did anyone find the solution?
`
"react": "16.6.3",
"react-native": "0.57.8",
"react-native-camera": "^2.11.0",
`

MateusAndrade, 2019-07-01:
@mg4u did you link your project follow https://react-native-community.github.io/react-native-camera/docs/installation?

mg4u, 2019-07-02:
@MateusAndrade I did what is mentioned in the docs `react-native link react-native-camera`,

but I will try the manual linking.

mg4u, 2019-07-02:
@MateusAndrade I have walked throw the steps and did all the same, but still, face the same exception.
<pre>
FAILURE: Build failed with an exception.

* Where:
Build file '/Users/mahmoudgamal/Documents/react/pos/android/build.gradle' line: 42

* What went wrong:
A problem occurred configuring project ':react-native-camera'.
> Failed to notify project evaluation listener.
   > Android tasks have already been created.
     This happens when calling android.applicationVariants,
     android.libraryVariants or android.testVariants.
     Once these methods are called, it is not possible to
     continue configuring the model.
</pre>

And that is my android/build.gradle line 42 section

<pre>

subprojects {
  project.configurations.all {
      afterEvaluate {project ->
        if (project.hasProperty("android")) {
            android {
                compileSdkVersion 27
                buildToolsVersion '27.0.3'
            }
        }
    }
  }
}
</pre>

the same version as mentioned in the docs.

MateusAndrade, 2019-07-06:
@mg4u you can provide the repo containing the code? Sorry for the late answer, i was busy on the last week and dont have time to look at Github.