react-native-camera/react-native-camera #2186: Can't build app with react-native-camera

gegham1, 2019-04-03:
### Versions
React Native: **0.55.4**
react-native-camera: **2.2.0**

```
dependencies {
    compile fileTree(dir: "libs", include: ["*.jar"])
    compile "com.android.support:appcompat-v7:23.0.1"
    compile 'com.android.support:exifinterface:25.+'
    compile "com.facebook.react:react-native:+"  // From node_modules
    implementation fileTree(dir: "libs", include: ["*.jar"])
    implementation 'com.android.support:appcompat-v7:26.0.1'
    implementation "com.facebook.react:react-native:+"  // From node_modules
    implementation project(':react-native-camera')
}
```

```
defaultConfig {
        applicationId "com.bonusapp"
        minSdkVersion 16
        targetSdkVersion 22
        versionCode 1
        versionName "1.0"
        ndk {
            abiFilters "armeabi-v7a", "x86"
        }
        missingDimensionStrategy 'react-native-camera', 'general' // <-- insert this line
    }
```

```
> Task :app:transformDexArchiveWithExternalLibsDexMergerForDebug FAILED
D8: Program type already present: com.facebook.infer.annotation.Assertions

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:transformDexArchiveWithExternalLibsDexMergerForDebug'.
> com.android.builder.dexing.DexArchiveMergerException: Error while merging dex archives: C:\Users\Gegham_Ayvazyan\Desktop\bonusApp\android\app\build\intermediates\transforms\dexBuilder\debug\0.jar, C:\Users\Gegham_Ayvazyan\Desktop\bonusApp\android\app\build\intermediates\transforms\dexBuilder\debug\1.jar, C:\Users\Gegham_Ayvazyan\Desktop\bonusApp\android\app\build\intermediates\transforms\dexBuilder\debug\2.jar, C:\Users\Gegham_Ayvazyan\Desktop\bonusApp\android\app\build\intermediates\transforms\dexBuilder\debug\3.jar, C:\Users\Gegham_Ayvazyan\Desktop\bonusApp\android\app\build\intermediates\transforms\dexBuilder\debug\8.jar, C:\Users\Gegham_Ayvazyan\Desktop\bonusApp\android\app\build\intermediates\transforms\dexBuilder\debug\9.jar, C:\Users\Gegham_Ayvazyan\Desktop\bonusApp\android\app\build\intermediates\transforms\dexBuilder\debug\17.jar, C:\Users\Gegham_Ayvazyan\Desktop\bonusApp\android\app\build\intermediates\transforms\dexBuilder\debug\18.jar, C:\Users\Gegham_Ayvazyan\Desktop\bonusApp\android\app\build\intermediates\transforms\dexBuilder\debug\19.jar, C:\Users\Gegham_Ayvazyan\Desktop\bonusApp\android\app\build\intermediates\transforms\dexBuilder\debug\20.jar, C:\Users\Gegham_Ayvazyan\Desktop\bonusApp\android\app\build\intermediates\transforms\dexBuilder\debug\21.jar, C:\Users\Gegham_Ayvazyan\Desktop\bonusApp\android\app\build\intermediates\transforms\dexBuilder\debug\22.jar, C:\Users\Gegham_Ayvazyan\Desktop\bonusApp\android\app\build\intermediates\transforms\dexBuilder\debug\23.jar, C:\Users\Gegham_Ayvazyan\Desktop\bonusApp\android\app\build\intermediates\transforms\dexBuilder\debug\24.jar, C:\Users\Gegham_Ayvazyan\Desktop\bonusApp\android\app\build\intermediates\transforms\dexBuilder\debug\25.jar, C:\Users\Gegham_Ayvazyan\Desktop\bonusApp\android\app\build\intermediates\transforms\dexBuilder\debug\26.jar, C:\Users\Gegham_Ayvazyan\Desktop\bonusApp\android\app\build\intermediates\transforms\dexBuilder\debug\27.jar, C:\Users\Gegham_Ayvazyan\Desktop\bonusApp\android\app\build\intermediates\transforms\dexBuilder\debug\28.jar, C:\Users\Gegham_Ayvazyan\Desktop\bonusApp\android\app\build\intermediates\transforms\dexBuilder\debug\30.jar, C:\Users\Gegham_Ayvazyan\Desktop\bonusApp\android\app\build\intermediates\transforms\dexBuilder\debug\31.jar, C:\Users\Gegham_Ayvazyan\Desktop\bonusApp\android\app\build\intermediates\transforms\dexBuilder\debug\32.jar, C:\Users\Gegham_Ayvazyan\Desktop\bonusApp\android\app\build\intermediates\transforms\dexBuilder\debug\33.jar, C:\Users\Gegham_Ayvazyan\Desktop\bonusApp\android\app\build\intermediates\transforms\dexBuilder\debug\35.jar, C:\Users\Gegham_Ayvazyan\Desktop\bonusApp\android\app\build\intermediates\transforms\dexBuilder\debug\36.jar, C:\Users\Gegham_Ayvazyan\Desktop\bonusApp\android\app\build\intermediates\transforms\dexBuilder\debug\37.jar, C:\Users\Gegham_Ayvazyan\Desktop\bonusApp\android\app\build\intermediates\transforms\dexBuilder\debug\38.jar, C:\Users\Gegham_Ayvazyan\Desktop\bonusApp\android\app\build\intermediates\transforms\dexBuilder\debug\39.jar, C:\Users\Gegham_Ayvazyan\Desktop\bonusApp\android\app\build\intermediates\transforms\dexBuilder\debug\40.jar, C:\Users\Gegham_Ayvazyan\Desktop\bonusApp\android\app\build\intermediates\transforms\dexBuilder\debug\52.jar, C:\Users\Gegham_Ayvazyan\Desktop\bonusApp\android\app\build\intermediates\transforms\dexBuilder\debug\53.jar, C:\Users\Gegham_Ayvazyan\Desktop\bonusApp\android\app\build\intermediates\transforms\dexBuilder\debug\54.jar, C:\Users\Gegham_Ayvazyan\Desktop\bonusApp\android\app\build\intermediates\transforms\dexBuilder\debug\55.jar, C:\Users\Gegham_Ayvazyan\Desktop\bonusApp\android\app\build\intermediates\transforms\dexBuilder\debug\56.jar
```

wladimir1982, 2019-04-03:
Have same issue

shtengel, 2019-04-04:
i was having the same issue, go to build.gradle of react-native-camera and change
`com.facebook.infer.annotation:+` to `com.facebook.infer.annotation:0.10.0`
it should work

DCKT, 2019-04-30:
I'm facing something similar to this issue, any update of this ? 

React Native: 0.59.2
react-native-camera: master

I have this message : 
```
* What went wrong:
Execution failed for task ':app:mergeDexDebug'.
> com.android.build.api.transform.TransformException: java.lang.RuntimeException: java.lang.RuntimeException: com.android.builder.dexing.DexArchiveMergerException: Error while merging dex archives: 
  The number of method references in a .dex file cannot exceed 64K.
  Learn how to resolve this issue at https://developer.android.com/tools/building/multidex.html
```

DCKT, 2019-05-02:
A little update : the module works perfectly on `1.13.1` version 🤔

liganghui, 2019-05-08:
> 我面临着与此问题类似的问题，对此有何更新？
> 
> React Native：0.59.2 
> react-native-camera：master
> 
> 我有这样的信息：
> 
> ```
> * What went wrong:
> Execution failed for task ':app:mergeDexDebug'.
> > com.android.build.api.transform.TransformException: java.lang.RuntimeException: java.lang.RuntimeException: com.android.builder.dexing.DexArchiveMergerException: Error while merging dex archives: 
>   The number of method references in a .dex file cannot exceed 64K.
>   Learn how to resolve this issue at https://developer.android.com/tools/building/multidex.html
> ```

I have the same problem, you can add the "multiDexEnabled" attribute to solve
in the android/app/build.gradle .   
```
android {

    defaultConfig {
        ...
        multiDexEnabled true   
    }
    ...
}
```

aidanw, 2019-05-14:
I add `multiDexEnabled true` The app now builds but when it deploys to my phone it crashes immediately.

I have tried other things here `https://developer.android.com/studio/build/multidex#java` and I cannot get the app to mount at all.

yqyily, 2019-05-16:
on app/build.gradle
defaultConfig{
multiDexEnabled true               //add this line
}
https://developer.android.com/studio/build/multidex

closed 2019-07-22 by stale[bot]

jhoanborges, 2019-10-28:
I had the same error. This is due the androidX integration. you need to downgrade to version 2.11.1