react-native-camera/react-native-camera #2191: Could not find method generalImplementation()

mbolukbasi, 2019-04-04:
# Bug Report
**To Do First**
- [ + ] Did you try latest release?
- [ + ] Did you try master?
- [ + ] Did you look for existing matching issues?

**Related Modules**
+RNCamera

**Platforms**
Android

**Versions**
- react-native-camera: 2.2.1
- react-native: 0.59.3

**Description/Current Behaviour**
Hi. Im new in react-native. I install react-native-camera on the clear build with the documentation steps then build failed.

* What went wrong:
A problem occurred evaluating project ':react-native-camera'.
> Could not find method generalImplementation() for arguments [com.google.android.gms:play-services-vision:17.0.2] on object of type org.gradle.api.internal.artifacts.dsl.dependencies.DefaultDependencyHandler.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 1s
error Could not install the app on the device, read the error above for details.
Make sure you have an Android emulator running or a device connected and have
set up your Android development environment:
https://facebook.github.io/react-native/docs/getting-started.html
error Command failed: gradlew.bat app:installDebug. Run CLI with --verbose flag for more details.

denizbas92, 2019-04-10:
I faced the same issue.. Any solution ?

mbolukbasi, 2019-04-15:
Yes. 
the problem is -generalImplementation and mlkitImplementation keywords in node_modules build.gradle file dependencies part.
Change with implementation.

denizbas92, 2019-04-15:
Now,this error happned.
 Required by:
         project :app
      > Cannot choose between the following variants of project :react-native-camera:
          - generalDebugRuntimeElements
          - mlkitDebugRuntimeElements
        All of them match the consumer attributes:
          - Variant 'generalDebugRuntimeElements':
              - Required com.android.build.api.attributes.BuildTypeAttr 'debug' and found compatible value 'debug'.
              - Found com.android.build.api.attributes.VariantAttr 'generalDebug' but wasn't required.
              - Required com.android.build.gradle.internal.dependency.AndroidTypeAttr 'Aar' and found compatible value 'Aar'.
              - Required org.gradle.usage 'java-runtime' and found compatible value 'java-runtime'.
              - Found react-native-camera 'general' but wasn't required.
          - Variant 'mlkitDebugRuntimeElements':
              - Required com.android.build.api.attributes.BuildTypeAttr 'debug' and found compatible value 'debug'.
              - Found com.android.build.api.attributes.VariantAttr 'mlkitDebug' but wasn't required.
              - Required com.android.build.gradle.internal.dependency.AndroidTypeAttr 'Aar' and found compatible value 'Aar'.
              - Required org.gradle.usage 'java-runtime' and found compatible value 'java-runtime'.
              - Found react-native-camera 'mlkit' but wasn't required.

denizbas92, 2019-04-15:
Now, an error  happened like this,
> java.nio.file.NoSuchFileException: /Users/apple/Desktop/ReactCamera/node_modules/react-native-camera/android/build/intermediates/merged_man?fests/generalDebug/../../l?brary_man?fest/generalDebug/AndroidManifest.xml

mbolukbasi, 2019-04-15:
your system is ios? my solution is for android.
try this in node_modules ->react-native-camera->android->build.gradle

dependencies {
    compileOnly 'com.facebook.react:react-native:+'
    compileOnly 'com.facebook.infer.annotation:infer-annotation:+'
    implementation "com.google.zxing:core:3.3.0"
    implementation "com.drewnoakes:metadata-extractor:2.9.1"
    implementation "com.google.android.gms:play-services-vision:${safeExtGet('googlePlayServicesVersion', '15.0.2')}"
    implementation "com.android.support:exifinterface:${safeExtGet('supportLibVersion', '27.1.0')}"
    implementation "com.android.support:support-annotations:${safeExtGet('supportLibVersion', '27.1.0')}"
    implementation "com.android.support:support-v4:${safeExtGet('supportLibVersion', '27.1.0')}"
}

denizbas92, 2019-04-16:
My system is android and versions are as shown below,
"react": "16.8.3",
"react-native": "0.59.4",
"react-native-camera": "^2.4.0"

erdemkosk, 2019-05-20:
any solution? same issue

Elbis, 2019-05-22:
same issue.

bsozeroglu, 2019-05-23:
it's working for me @mbolukbasi Thanks :)

mrkacan, 2019-06-09:
Same issue.

closed 2019-08-15 by stale[bot]

nurikon, 2019-09-01:
my Turkish friends. it is not a coincidence that everyone is Turkish i this page. if your coputer language is turkish, change capital ''I'' to ''İ'' in react-native-camera build.gradel

generalImplementation>>generalİmplementation
mlkitImplementation>>mlkitİmplementation
 
and it is solved ;)

abltcy, 2019-11-01:
Tanks @nurikon

fekajin, 2019-11-15:
thanks @nurikon

ErtugrulBEKTIK, 2019-12-19:
Thank you my Turkish brother @nurikon

aykutceylan, 2020-01-08:
> 
> 
> my Turkish friends. it is not a coincidence that everyone is Turkish i this page. if your coputer language is turkish, change capital ''I'' to ''İ'' in react-native-camera build.gradel
> 
> generalImplementation>>generalİmplementation
> mlkitImplementation>>mlkitİmplementation
> 
> and it is solved ;)

Thank You Turkish Bro :) @nurikon 

Is there any permanent solution for this problem?

dilankara, 2020-02-07:
> Tanks @nurikon

Thanks :)

dilankara, 2020-02-07:
> my Turkish friends. it is not a coincidence that everyone is Turkish i this page. if your coputer language is turkish, change capital ''I'' to ''İ'' in react-native-camera build.gradel
> 
> generalImplementation>>generalİmplementation
> mlkitImplementation>>mlkitİmplementation
> 
> and it is solved ;)

Thanks :)

alperenarc, 2020-03-08:
> my Turkish friends. it is not a coincidence that everyone is Turkish i this page. if your coputer language is turkish, change capital ''I'' to ''İ'' in react-native-camera build.gradel
> 
> generalImplementation>>generalİmplementation
> mlkitImplementation>>mlkitİmplementation
> 
> and it is solved ;)

It's worked. What a solution. It was very interesting to me. It was the first time that I saw a solution specific to the Turks. :

alisaracoglu, 2020-04-02:
thanks @nurikon, a turkish-only issue =))