react-native-camera/react-native-camera #2200: Base64 option is not working in latest release 2.2.2

UmaNallasamy, 2019-04-10:
# Bug Report
**To Do First**
- [ ] Did you try latest release? - Yes
- [ ] Did you try master? - No
- [ ] Did you look for existing matching issues? - Yes

**Related Modules**
Issue is with RNCamera

**Platforms**
Both Android and IOS

**Versions**

- Android:
- iOS:
- react-native-camera:2.2.2

**Description/Current Behaviour**
I have passed base64:true in the options but returned promise object does not have base64 key.
code snippet:
=========
const options = {quality: 0.5, base64: true,fixOrientation: true,forceUpOrientation:true,skipProcessing: false};
                const data = await this.camera.takePictureAsync(options)
				alert(Object.keys(data))
.......................................
 <RNCamera
                    ref={ref => {
                        this.camera = ref;
                    }}
                    style={preview}
                    type={this.state.switchCamera ? RNCamera.Constants.Type.back : RNCamera.Constants.Type.front}
                    flashMode={RNCamera.Constants.FlashMode.off}
                    permissionDialogTitle={'Permission to use camera'}
                    permissionDialogMessage={'We need your permission to use your camera phone'}
                />

**Expected Behaviour**
I was trying to access data.base64  in returned object but this key is not present.

**Steps to Reproduce**

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

dov11, 2019-04-14:
For a quick fix try removing option `skipProcessing: false`. When you set any value to this key, it [seems](https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/org/reactnative/camera/tasks/ResolveTakenPictureAsyncTask.java#L57) to be automatically treated as `true`.

UmaNallasamy, 2019-04-15:
Thanks, it works. Will this bug "skipProcessing" flag be resolved in next release?

closed 2019-06-21 by stale[bot]