react-native-camera/react-native-camera #2201: pictureSize does't change when ratio changed

hushicai, 2019-04-10:
# Bug Report
react native: 0.48.4
react-native-camera: 1.11.1
modules: Camera1

## Steps

As mentioned in #1280 , I want to choose the best ratio when onCameraReady.

For example, the initial ratio is '4:3', then i change to `16:9`, but i found that the pictureSize ratio doesn't change at all.

after some debug, i found the problem may be [here](https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera1.java#L659).

`mPictureSize` is not null when the second call of `adjustCameraParameters`.

[image]

As we can see above, the pictureSize remains 4032x3024, which ratio is `4032/3024=4/3`, not match the latest ratio `16:9`.

hushicai, 2019-04-10:
Is there really a problem, or is it that I am not using it properly?
Can everyone give me some advice?

hushicai, 2019-04-10:
@sibelius Thanks for looking at this. can I just remove the if condition `mPictureSize == null` and always get the latest pictureSizes with the latest AspectRatio every time `adjustCameraParameters` called?

sibelius, 2019-04-10:
I think this makes sense

hushicai, 2019-04-10:
Thanks, I will patch it as a temporary solution:

```diff
--- a/android/src/main/java/com/google/android/cameraview/Camera1.java
+++ b/android/src/main/java/com/google/android/cameraview/Camera1.java
@@ -657,9 +657,7 @@ class Camera1 extends CameraViewImpl implements MediaRecorder.OnInfoListener,
         Size size = chooseOptimalSize(sizes);
 
         // Always re-apply camera parameters
-        if (mPictureSize == null) {
-            mPictureSize = mPictureSizes.sizes(mAspectRatio).last();
-        }
+        mPictureSize = mPictureSizes.sizes(mAspectRatio).last();
         if (mShowingPreview) {
             mCamera.stopPreview();
             mIsPreviewActive = false;
```

sibelius, 2019-04-10:
make sure this does not cause any regression

and then send a pull request fixing it

hushicai, 2019-04-12:
closes with #2204

closed 2019-04-12 by hushicai