react-native-camera/react-native-camera #2202: Crash when switching from back camera to front camera

cblavier, 2019-04-10:
# Bug Report
**To Do First**
- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Related Modules**
RNCamera

**Platforms**
iOS

**Versions**
- iOS: 12.1.4
- react-native-camera: 2.2.0
- react-native: 0.57
- react: 16.5.0

**Description/Current Behaviour**
On some iPhone models (iphone 7, iphone XS) switching from back to front camera after a first recording session make the camera crash.

It's possible to switch back and forth between cameras before recording, but when recording has started with the back camera, if you stop recording and try to switch to front camera, the camera component will crash (black screen)

**Expected Behaviour**
We expect the camera not to crash!

**Steps to Reproduce**
- clone this [project](https://github.com/phenixdigital/switch-camera-issue)
- run it on iphone7 or later (it doesn't crash on iphone 6)
- start recording
- stop recording
- try to switch camera
- crash: black screen

**Does it work with Expo Camera?**
Yes, it is working properly with Expo 32.0.0

jasoncoopercovideo, 2019-05-02:
We were experiencing this issue as well.  However we did find a very simple, temporary workaround.  We're still not sure why it works, but it does.  The switching is not as fast as when it is not recording, but still suffices.  We found the fix on StackOverflow but I can't find it anymore.

We just assigned a key attribute to the <RNCamera> Element, then update the key value when switching the camera while recording.  It's forcing a re-render, but it should be re-rendering anyway, so I still don't know why it works.

Pseudocode Example below.  Hope this helps!

`
switchCam() {
  if( isRecording) {
    this.setState({
      type: this.state.type === 'front' ? 'back' : 'front',
      recordingKey: Math.random()
    });
  }
 else {
    this.setState({
      type: this.state.type === 'front' ? 'back' : 'front',
    });
 }
}

<RNCamera
  key={this.state.recordingKey}
  type={this.state.camType}
  .../>
`

closed 2019-07-08 by stale[bot]

jaroslav009, 2021-03-31:
any answers?