react-native-camera/react-native-camera #2203: Unfortunately, app has stopped

deinega, 2019-04-10:
I followed all the installation instructions for Android, and now my app crashes at start: "Unfortunately, app has stopped"

I runned `adb logcat`, and this is what I found in log messages:

04-10 10:03:56.628 17959 17959 E AndroidRuntime: java.lang.NoClassDefFoundError: Failed resolution of: Lcom/google/android/gms/common/internal/zzbq;
04-10 10:03:56.628 17959 17959 E AndroidRuntime:        at com.google.firebase.provider.FirebaseInitProvider.attachInfo(Unknown Source)
04-10 10:03:56.628 17959 17959 E AndroidRuntime:        at android.app.ActivityThread.installProvider(ActivityThread.java:5175)
04-10 10:03:56.628 17959 17959 E AndroidRuntime:        at android.app.ActivityThread.installContentProviders(ActivityThread.java:4770)
04-10 10:03:56.628 17959 17959 E AndroidRuntime:        at android.app.ActivityThread.handleBindApplication(ActivityThread.java:4710)
04-10 10:03:56.628 17959 17959 E AndroidRuntime:        at android.app.ActivityThread.access$1600(ActivityThread.java:155)
04-10 10:03:56.628 17959 17959 E AndroidRuntime:        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1410)
04-10 10:03:56.628 17959 17959 E AndroidRuntime:        at android.os.Handler.dispatchMessage(Handler.java:102)
04-10 10:03:56.628 17959 17959 E AndroidRuntime:        at android.os.Looper.loop(Looper.java:174)
04-10 10:03:56.628 17959 17959 E AndroidRuntime:        at android.app.ActivityThread.main(ActivityThread.java:5440)
04-10 10:03:56.628 17959 17959 E AndroidRuntime:        at java.lang.reflect.Method.invoke(Native Method)
04-10 10:03:56.628 17959 17959 E AndroidRuntime:        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
04-10 10:03:56.628 17959 17959 E AndroidRuntime:        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
04-10 10:03:56.628 17959 17959 E AndroidRuntime: Caused by: java.lang.ClassNotFoundException: Didn't find class "com.google.android.gms.common.internal.zzbq"

How can I resolve this issue ? 
Thank you!

sibelius, 2019-04-10:
the bug is in your com.google.android.gms packages

there is more than one or it is missing

deinega, 2019-04-10:
This is the fresh react-native app with couple of react-native packages installed...

Where I can check if I have more that one or missing com.google.android.gms package?

closed 2019-06-16 by stale[bot]

pzehle, 2021-04-13:
Even though this is an old issue, on a clean install of `react-native-camera`, on Android, this error starts again. After 4 hours of debugging and trying everything, I found out that you need to import Firebase in your project, so even though the docs clearly say `Firebase is not required`, as of today, Firebase is in fact required, at least for Android to work with this library.

What I did was just to import:

`implementation platform('com.google.firebase:firebase-bom:27.0.0')`

IMO firebase is completely destroyed right now, so I just avoid using it as much as I can, but it seems that this library requires it anyway...