react-native-camera/react-native-camera #2210: Switch to newer version, build fails

tapir, 2019-04-16:
# Question

**Ask your Question**
I've switched my app from `react-native-camera` version 1.6.4 to 1.13.1
I've also switched from gradle 4.6 to 4.10 (if this makes any difference)

and `./gradlew assembleDebug` now fails with these errors
```
Could not resolve all files for configuration ':react-native-camera:debugCompileClasspath'.
> Could not resolve com.android.support:support-compat:27.1.1.
  Required by:
      project :react-native-camera
   > Cannot find a version of 'com.android.support:support-compat' that satisfies the version constraints: 
        Dependency path 'myApp:react-native-camera:unspecified' --> 'com.android.support:support-v4:27.1.1' --> 'com.android.support:support-compat:27.1.1'
        Dependency path 'myApp:react-native-camera:unspecified' --> 'com.facebook.react:react-native:0.59.4' --> 'com.android.support:appcompat-v7:28.0.0' --> 'com.android.support:support-compat:28.0.0'
        Dependency path 'myApp:react-native-camera:unspecified' --> 'com.android.support:support-v4:27.1.1' --> 'com.android.support:support-media-compat:27.1.1' --> 'com.android.support:support-compat:27.1.1'
        Dependency path 'myApp:react-native-camera:unspecified' --> 'com.android.support:support-v4:27.1.1' --> 'com.android.support:support-core-utils:28.0.0' --> 'com.android.support:support-compat:28.0.0'
        Dependency path 'myApp:react-native-camera:unspecified' --> 'com.android.support:support-v4:27.1.1' --> 'com.android.support:support-fragment:28.0.0' --> 'com.android.support:support-compat:28.0.0'
        Dependency path 'myApp:react-native-camera:unspecified' --> 'com.facebook.react:react-native:0.59.4' --> 'com.android.support:appcompat-v7:28.0.0' --> 'com.android.support:support-vector-drawable:28.0.0' --> 'com.android.support:support-compat:28.0.0'
        Dependency path 'myApp:react-native-camera:unspecified' --> 'com.google.android.gms:play-services-vision:15.0.2' --> 'com.google.android.gms:play-services-basement:15.0.1' --> 'com.android.support:support-v4:27.1.1' --> 'com.android.support:support-compat:27.1.1'
        Dependency path 'myApp:react-native-camera:unspecified' --> 'com.android.support:support-v4:27.1.1' --> 'com.android.support:support-core-utils:28.0.0' --> 'com.android.support:loader:28.0.0' --> 'com.android.support:support-compat:28.0.0'
        Dependency path 'myApp:react-native-camera:unspecified' --> 'com.android.support:support-v4:27.1.1' --> 'com.android.support:support-core-ui:28.0.0' --> 'com.android.support:support-compat:28.0.0'
        Dependency path 'myApp:react-native-camera:unspecified' --> 'com.android.support:support-v4:27.1.1' --> 'com.android.support:support-core-ui:28.0.0' --> 'com.android.support:customview:28.0.0' --> 'com.android.support:support-compat:28.0.0'
        Dependency path 'myApp:react-native-camera:unspecified' --> 'com.android.support:support-v4:27.1.1' --> 'com.android.support:support-core-ui:28.0.0' --> 'com.android.support:viewpager:28.0.0' --> 'com.android.support:support-compat:28.0.0'
        Dependency path 'myApp:react-native-camera:unspecified' --> 'com.android.support:support-v4:27.1.1' --> 'com.android.support:support-core-ui:28.0.0' --> 'com.android.support:coordinatorlayout:28.0.0' --> 'com.android.support:support-compat:28.0.0'
        Dependency path 'myApp:react-native-camera:unspecified' --> 'com.android.support:support-v4:27.1.1' --> 'com.android.support:support-core-ui:28.0.0' --> 'com.android.support:drawerlayout:28.0.0' --> 'com.android.support:support-compat:28.0.0'
        Dependency path 'myApp:react-native-camera:unspecified' --> 'com.android.support:support-v4:27.1.1' --> 'com.android.support:support-core-ui:28.0.0' --> 'com.android.support:slidingpanelayout:28.0.0' --> 'com.android.support:support-compat:28.0.0'
        Dependency path 'myApp:react-native-camera:unspecified' --> 'com.android.support:support-v4:27.1.1' --> 'com.android.support:support-core-ui:28.0.0' --> 'com.android.support:swiperefreshlayout:28.0.0' --> 'com.android.support:support-compat:28.0.0'
        Dependency path 'myApp:react-native-camera:unspecified' --> 'com.android.support:support-v4:27.1.1' --> 'com.android.support:support-core-ui:28.0.0' --> 'com.android.support:asynclayoutinflater:28.0.0' --> 'com.android.support:support-compat:28.0.0'
        Constraint path 'myApp:react-native-camera:unspecified' --> 'com.android.support:support-compat' strictly '27.1.1' because of the following reason: debugRuntimeClasspath uses version 27.1.1
        Constraint path 'myApp:react-native-camera:unspecified' --> 'com.android.support:support-compat' strictly '27.1.1' because of the following reason: debugRuntimeClasspath uses version 27.1.1
        Constraint path 'myApp:react-native-camera:unspecified' --> 'com.android.support:support-compat' strictly '27.1.1' because of the following reason: debugRuntimeClasspath uses version 27.1.1
        Constraint path 'myApp:react-native-camera:unspecified' --> 'com.android.support:support-compat' strictly '27.1.1' because of the following reason: debugRuntimeClasspath uses version 27.1.1

> Could not resolve com.android.support:support-annotations:27.1.1.
  Required by:
      project :react-native-camera
   > Cannot find a version of 'com.android.support:support-annotations' that satisfies the version constraints: 
        Dependency path 'myApp:react-native-camera:unspecified' --> 'com.android.support:support-annotations:27.1.1'
        Constraint path 'myApp:react-native-camera:unspecified' --> 'com.android.support:support-annotations' strictly '27.1.1' because of the following reason: debugRuntimeClasspath uses version 27.1.1
        Dependency path 'myApp:react-native-camera:unspecified' --> 'com.android.support:support-annotations:27.1.1'
        Constraint path 'myApp:react-native-camera:unspecified' --> 'com.android.support:support-annotations' strictly '27.1.1' because of the following reason: debugRuntimeClasspath uses version 27.1.1
        Dependency path 'myApp:react-native-camera:unspecified' --> 'com.android.support:support-annotations:27.1.1'
        Constraint path 'myApp:react-native-camera:unspecified' --> 'com.android.support:support-annotations' strictly '27.1.1' because of the following reason: debugRuntimeClasspath uses version 27.1.1
        Dependency path 'myApp:react-native-camera:unspecified' --> 'com.android.support:support-annotations:27.1.1'
        Constraint path 'myApp:react-native-camera:unspecified' --> 'com.android.support:support-annotations' strictly '27.1.1' because of the following reason: debugRuntimeClasspath uses version 27.1.1
        Dependency path 'myApp:react-native-camera:unspecified' --> 'com.android.support:support-annotations:27.1.1'
        Constraint path 'myApp:react-native-camera:unspecified' --> 'com.android.support:support-annotations' strictly '27.1.1' because of the following reason: debugRuntimeClasspath uses version 27.1.1
        Dependency path 'myApp:react-native-camera:unspecified' --> 'com.android.support:support-annotations:27.1.1'
        Dependency path 'myApp:react-native-camera:unspecified' --> 'com.android.support:exifinterface:27.1.1' --> 'com.android.support:support-annotations:27.1.1'
        Dependency path 'myApp:react-native-camera:unspecified' --> 'com.facebook.react:react-native:0.59.4' --> 'com.android.support:appcompat-v7:28.0.0' --> 'com.android.support:support-annotations:28.0.0'
        Dependency path 'myApp:react-native-camera:unspecified' --> 'com.android.support:support-v4:27.1.1' --> 'com.android.support:support-media-compat:27.1.1' --> 'com.android.support:support-annotations:27.1.1'
        Dependency path 'myApp:react-native-camera:unspecified' --> 'com.android.support:support-v4:27.1.1' --> 'com.android.support:support-compat:28.0.0' --> 'com.android.support:support-annotations:28.0.0'
        Dependency path 'myApp:react-native-camera:unspecified' --> 'com.android.support:support-v4:27.1.1' --> 'com.android.support:support-core-utils:28.0.0' --> 'com.android.support:support-annotations:28.0.0'
        Dependency path 'myApp:react-native-camera:unspecified' --> 'com.android.support:support-v4:27.1.1' --> 'com.android.support:support-fragment:28.0.0' --> 'com.android.support:support-annotations:28.0.0'
        Dependency path 'myApp:react-native-camera:unspecified' --> 'com.facebook.react:react-native:0.59.4' --> 'com.android.support:appcompat-v7:28.0.0' --> 'com.android.support:collections:28.0.0' --> 'com.android.support:support-annotat [... cut ...]
        Dependency path 'myApp:react-native-camera:unspecified' --> 'com.android.support:support-v4:27.1.1' --> 'com.android.support:support-core-utils:28.0.0' --> 'com.android.support:loader:28.0.0' --> 'android.arch.lifecycle:livedata:1.1.1' --> 'android.arch.lifecycle:livedata-core:1.1.1'
        Constraint path 'myApp:react-native-camera:unspecified' --> 'android.arch.lifecycle:livedata-core' strictly '1.1.0' because of the following reason: debugRuntimeClasspath uses version 1.1.0
```

Any idea how I can make build work again?

Linoa65, 2019-04-16:
Same here ! But I just updated the gradle version and not the rn-camera module. I think it's an issue with gradle

Linoa65, 2019-04-16:
I just switch to latest release, then add the following to my android/app/build.gradle

```gradle
android {
  ...
  defaultConfig {
    ...
    missingDimensionStrategy 'react-native-camera', 'general' // <-- insert this line
  }
}
```

Feel free to close if it works for you !

sibelius, 2019-04-16:
follow migration guide to v2

tapir, 2019-04-16:
> follow migration guide to v2

I'm not on version 2. It's 1.13.1

tapir, 2019-04-16:
> I just switch to latest release, then add the following to my android/app/build.gradle
> 
> ```groovy-gradle
> android {
>   ...
>   defaultConfig {
>     ...
>     missingDimensionStrategy 'react-native-camera', 'general' // <-- insert this line
>   }
> }
> ```
> 
> Feel free to close if it works for you !

didn't work for me. Can you tell which version you've switched to exactly?

Linoa65, 2019-04-16:
I'm using react-native-camera 2.4.0 :)

tapir, 2019-04-16:
thanks that version fixes the problems.

@sibelius you can close if you like. I'm pretty sure no dev will care for 1.x bugs.

closed 2019-04-16 by sibelius