react-native-camera/react-native-camera #2211: The key to cropping the preview image! please add to docs!!

timomeara, 2019-04-16:
# Feature-Request

**Describe the Feature**
Please document this.
there are many many issues/questions here regarding cropping or displaying an arbitrarily sized camera view. 
after 2 days of chasing my tail, i finally found this;
add **overflow: 'hidden'** to the view containing the  `<RNCamera/>`

(from #2146 )
```
  <View style={ styles.preview }>\
    <RNCamera
      ref={ cam => {this.camera = cam } }
      style={ styles.camera }
      type={ RNCamera.Constants.Type.front }
      flashMode={ RNCamera.Constants.FlashMode.on }
      ratio={ '16:9' }
    />
  </View>
```
```
 preview: {
    height: 240,
    width: 240,
    overflow: 'hidden' 
  },
  camera: {
    height: 240,
    width: 240
  }
```

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this feature to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

sibelius, 2019-04-16:
feel free to improve docs

EskelCz, 2019-06-03:
I've tried the overflow:hidden trick and it crashed the app, on android.

closed 2019-08-09 by stale[bot]