react-native-camera/react-native-camera #2212: provide a real-life use case example

ppaul, 2019-04-17:
# Question
Could you please provide a detached example, in which you just use the library as anyone would use it in their project? I tried for 2 days but could not manage to get it working along with a firebase auth starter project. The existing basic example use the library from upper folder, which is not the case for all real life projects where it's used.
Thank you

sibelius, 2019-04-17:
feel free to create one and link to the docs

the only difference is how we manage our metro.config.js so we don't have duplicated package on node_modules

MateusAndrade, 2019-04-22:
I used this project on my daily job. I think i got a private repo, using this library. i can send url to attach this on the docs.

ppaul, 2019-04-26:
it's OK now. I was struggling a lot mainly because some libs get updated and I had errors which people didn't got on stackoverflow at the moment code was released

closed 2019-04-26 by ppaul