react-native-camera/react-native-camera #2219: In Android environment, I switched on the flashlight at the same time when I scanned the code and sent the wrong message.

ymw2017, 2019-04-19:
The error log is as follows:
camera_in_use(4):setTorchMode:1466:Torch for camera "0" is not available due to an existing camera user

How can I turn on the flashlight while sweeping the code?Thank you.

alizorlu, 2019-04-20:
> The error log is as follows:
> camera_in_use(4):setTorchMode:1466:Torch for camera "0" is not available due to an existing camera user
> 
> How can I turn on the flashlight while sweeping the code?Thank you.

[https://github.com/react-native-community/react-native-camera/issues/2220#issuecomment-485138836](url) might help you

closed 2019-06-26 by stale[bot]

ymw2017, 2019-06-27:
@alizorlu  Thank you very much for helping me solve this problem.

Macieyou, 2023-02-19:
> @alizorlu Thank you very much for helping me solve this problem.

Hi @ymw2017 
I have the same problem. How exactly did you solve this?