react-native-camera/react-native-camera #2220: Android app crashed when turn on-off camera flash light.

akashmi, 2019-04-20:
# Bug Report

Android app is crashing when trying
[image]
 to turn on-off camera flash in my one plush5 device.
It is working fine in iOS and other Android devices.

alizorlu, 2019-04-20:
Heyy , RNCamera ,view code ? take a screen shot and add there please

akashmi, 2019-04-20:
```
 <RNCamera
                        ref={ref => {
                            this.camera = ref;
                        }}
                        style={{
                            flex: 1,
                            backgroundColor: '#EBEBEB'
                        }}
                        type={this.state.cameraMode}
                        flashMode={this.state.flashMode}
                    />

```

alizorlu, 2019-04-20:
> Here is RNCamera view code
> 
> ```
> <RNCamera
>                        ref={ref => {
>                            this.camera = ref;
>                        }}
>                        style={{
>                            flex: 1,
>                            backgroundColor: '#EBEBEB'
>                        }}
>                        type={this.state.cameraMode}
>                        flashMode={this.state.flashMode}
>                    />
> ```

[image]

would you try again image suggestion @akashmi

akashmi, 2019-06-20:
> > Here is RNCamera view code
> > ```
> > <RNCamera
> >                        ref={ref => {
> >                            this.camera = ref;
> >                        }}
> >                        style={{
> >                            flex: 1,
> >                            backgroundColor: '#EBEBEB'
> >                        }}
> >                        type={this.state.cameraMode}
> >                        flashMode={this.state.flashMode}
> >                    />
> > ```
> 
> [image]
> 
> would you try again image suggestion @akashmi
> 
> **Note : RNCamera <= import statement**

@alizorlu Yes. I tried it but still getting the same issue.

MateusAndrade, 2019-06-22:
@akashmi you can give a try with the example app? Please give a look at: https://github.com/react-native-community/react-native-camera/tree/master/examples/basic

akashmi, 2019-08-21:
@MateusAndrade I tried it. still getting the same issue in Motorola devices. it is working fine in oneplush device.

closed 2019-10-27 by stale[bot]

anujraghuvanshi, 2022-03-23:
Still getting in 2022. Anybody with possible reasons ?