react-native-camera/react-native-camera #2221: v2.6.0 face detection doesn't work

michaelcurtis, 2019-04-20:
# Bug Report

**To Do First**

- [ Y ] Did you try latest release?
- [ Y ] Did you try master?
- [ Y ] Did you look for existing matching issues?

**Platforms**

iOS

**Versions**

- XCode: 10.2 (10E125)
- iOS: 12.2
- react-native-camera: 2.6.0 (tried master too)
- react-native: 0.59.5
- react: 16.8.3
  

**Description/Current Behaviour**

The onFacesDetected callback is fired endlessly with an empty faces object. The value looks like this: { type: 'face', faces: [], target: 23 }. It does this whether or not faces are in the camera field of view. 

**Expected Behaviour**

The onFacesDetected callback to be fired: 1. when faces are detected and 2. with a populated faces object.

**Steps to Reproduce**

1. Create new react native app
2. yarn add react-native-camera@2.6.0
3. Update Podfile
4. pod install
5. Set up Firebase (confirmed it connected in Firebase when the app runs)
6. Compile and run

***Podfile***
```
target 'CameraTest' do
  pod 'React', :path => '../node_modules/react-native', :subspecs => [
    'Core',
    'CxxBridge',
    'DevSupport',
    'RCTText',
    'RCTNetwork',
    'RCTWebSocket',
    'RCTAnimation',
  ]

  pod 'yoga', :path => '../node_modules/react-native/ReactCommon/yoga'
  pod 'DoubleConversion', :podspec => '../node_modules/react-native/third-party-podspecs/DoubleConversion.podspec'
  pod 'glog', :podspec => '../node_modules/react-native/third-party-podspecs/glog.podspec'
  pod 'Folly', :podspec => '../node_modules/react-native/third-party-podspecs/Folly.podspec'

  pod 'Firebase/Core'
  pod 'react-native-camera', path: '../node_modules/react-native-camera', subspecs: [
    'FaceDetectorMLKit'
  ]

  target 'CameraTestTests' do
    inherit! :search_paths
  end
end

post_install do |installer|
  installer.pods_project.targets.each do |target|
    if target.name == "React"
      target.remove_from_project
    end
  end
end
```

***AppDelegate.m***
```
#import "AppDelegate.h"

#import <React/RCTBridge.h>
#import <React/RCTBundleURLProvider.h>
#import <React/RCTRootView.h>

@import Firebase;

@implementation AppDelegate

- (BOOL)application:(UIApplication *)application didFinishLaunchingWithOptions:(NSDictionary *)launchOptions
{
  RCTBridge *bridge = [[RCTBridge alloc] initWithDelegate:self launchOptions:launchOptions];
  RCTRootView *rootView = [[RCTRootView alloc] initWithBridge:bridge
                                                   moduleName:@"CameraTest"
                                            initialProperties:nil];
  
  rootView.backgroundColor = [[UIColor alloc] initWithRed:1.0f green:1.0f blue:1.0f alpha:1];

  self.window = [[UIWindow alloc] initWithFrame:[UIScreen mainScreen].bounds];
  UIViewController *rootViewController = [UIViewController new];
  rootViewController.view = rootView;
  self.window.rootViewController = rootViewController;
  [self.window makeKeyAndVisible];
  [FIRApp configure];
  return YES;
}

- (NSURL *)sourceURLForBridge:(RCTBridge *)bridge
{
#if DEBUG
  return [[RCTBundleURLProvider sharedSettings] jsBundleURLForBundleRoot:@"index" fallbackResource:nil];
#else
  return [[NSBundle mainBundle] URLForResource:@"main" withExtension:@"jsbundle"];
#endif
}

@end
```

***App.js***
Note: I tested both the front and back camera, neither one works.
```
import React from 'react';
import { RNCamera } from 'react-native-camera';

export default class App extends React.Component {
  onFacesDetected = (faces) => {
    console.log(faces);
  }

  render() {
    return (
      <RNCamera
        style={{ width: '100%', height: '100%' }}
        type={RNCamera.Constants.Type.front}
        onFacesDetected={this.onFacesDetected}
        faceDetectionMode={RNCamera.Constants.FaceDetection.Mode.fast}
        faceDetectionLandmarks={RNCamera.Constants.FaceDetection.Landmarks.all}
        faceDetectionClassifications={RNCamera.Constants.FaceDetection.Classifications.all}
      />
    );
  }
}
```

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

sibelius, 2019-04-20:
in which version does it worked?

michaelcurtis, 2019-04-20:
Thanks for such a quick reply. I haven't been able to get any of the mlkit versions working. The last one I have working is 2.1.1 with manually linked GoogleMobileVision libraries. Let me know if any more info would be helpful.

sibelius, 2019-04-20:
can you try this example https://github.com/react-native-community/react-native-camera/tree/master/examples/mlkit

can you give more info about the issue?

michaelcurtis, 2019-04-20:
I try to run the example and circle back.

Regarding the issue. What I see is the camera preview running as expected. 
The onFacesDetected prop is called every half second or so with an empty faces object. 
It is called regardless of whether there is a face in the field of view. 

Some log lines: 
2019-04-20 13:36:34.835014-0700 CameraTest[35012:3234542] [BoringSSL] nw_protocol_boringssl_get_output_frames(1301) [C15.1:2][0x109d0cb50] get output frames failed, state 8196
2019-04-20 13:36:34.835073-0700 CameraTest[35012:3234542] [BoringSSL] nw_protocol_boringssl_get_output_frames(1301) [C15.1:2][0x109d0cb50] get output frames failed, state 8196
2019-04-20 13:36:34.835189-0700 CameraTest[35012:3234542] TIC Read Status [15:0x0]: 1:57
2019-04-20 13:36:34.835200-0700 CameraTest[35012:3234542] TIC Read Status [15:0x0]: 1:57
2019-04-20 13:36:34.842 [info][tid:com.facebook.react.JavaScript] { type: 'face', faces: [], target: 15 }
2019-04-20 13:36:34.842056-0700 CameraTest[35012:3234537] { type: 'face', faces: [], target: 15 }
2019-04-20 13:36:35.422 [info][tid:com.facebook.react.JavaScript] { type: 'face', faces: [], target: 15 }
2019-04-20 13:36:35.422517-0700 CameraTest[35012:3234537] { type: 'face', faces: [], target: 15 }
2019-04-20 13:36:35.988 [info][tid:com.facebook.react.JavaScript] { type: 'face', faces: [], target: 15 }
2019-04-20 13:36:35.988249-0700 CameraTest[35012:3234537] { type: 'face', faces: [], target: 15 }
2019-04-20 13:36:36.554 [info][tid:com.facebook.react.JavaScript] { type: 'face', faces: [], target: 15 }
2019-04-20 13:36:36.554499-0700 CameraTest[35012:3234537] { type: 'face', faces: [], target: 15 }

michaelcurtis, 2019-04-20:
I got the mlkit example project running. Same issue. I see the preview and all the example project settings / buttons. I press "detect faces" so that it reads "detecting faces". Then onFacesDetected is called with an empty face object every half second or so, regardless if there is a face in camera field of view. Here is the output from react-native info, if helpful. 

```
$ react-native info
info 
  React Native Environment Info:
    System:
      OS: macOS 10.14.4
      CPU: (8) x64 Intel(R) Core(TM) i7-6700HQ CPU @ 2.60GHz
      Memory: 222.45 MB / 16.00 GB
      Shell: 3.2.57 - /bin/bash
    Binaries:
      Node: 11.14.0 - /usr/local/bin/node
      Yarn: 1.15.2 - /usr/local/bin/yarn
      npm: 5.6.0 - /usr/local/bin/npm
      Watchman: 4.9.0 - /usr/local/bin/watchman
    SDKs:
      iOS SDK:
        Platforms: iOS 12.2, macOS 10.14, tvOS 12.2, watchOS 5.2
    IDEs:
      Android Studio: 3.2 AI-181.5540.7.32.5056338
      Xcode: 10.2/10E125 - /usr/bin/xcodebuild
    npmPackages:
      react: 16.8.3 => 16.8.3 
      react-native: 0.59.1 => 0.59.1 
```

**Aside**
I had to do additional steps to make the example project work: 
1. yarn add @babel/runtime@7.0.0-beta.55
2. cd ios && pod install
3. cd ../../ && yarn install

sibelius, 2019-04-21:
does it work on android?

does it work with back camera on ios?

michaelcurtis, 2019-04-21:
Doesn't work with back or front camera on iOS. 

I'll set it up on Android and let you know.

michaelcurtis, 2019-04-21:
Android back camera: works perfectly. 
Android front camera: onFacesDetected is called with populated faces object, but inconsistently and with incorrect coordinates.

Google Pixel 3
Android version: 9
Build number: PQ2A.190205.001

sibelius, 2019-04-21:
check video preset

master worked for me on iPhone 6

https://github.com/react-native-community/react-native-camera/pull/2044
https://github.com/react-native-community/react-native-camera/pull/2036

michaelcurtis, 2019-04-21:
I have all of the changes committed in #2193.
I tried manually applying the changes in #2044 and then #2036 individually. Neither works.

I have confirmed that I am compiling the RNCamera my mlkit example app is using from source.

I'll try some more tweaks in RNCamera.m to see if I can get it working. But I'm not sure this issue is related to video preset.

michaelcurtis, 2019-04-22:
I have this narrowed down to `RNCameraUtils convertBufferToUIImage`, though I don't know what it's doing in creating the UIImage that the face detector doesn't like. (Side note: text detection works correctly on front and back camera).

I got face detection working by creating a FIRVisionImage directly from the camera buffer then handing that FIRVisionImage into a modified `FaceDetectorManagerMlkit findFacesInFrame` that takes a FIRVisionImage rather than a UIImage. Sourced that technique from [Google's example app](https://github.com/firebase/quickstart-ios/tree/master/mlvision/MLVisionExampleObjc).

```
CVImageBufferRef imageBuffer = CMSampleBufferGetImageBuffer(sampleBuffer);
FIRVisionImage *visionImage = [[FIRVisionImage alloc] initWithBuffer:sampleBuffer];
```

sibelius, 2019-04-23:
Are you going to send a PR fixing the issue?

michaelcurtis, 2019-04-23:
I'll try. Basically learning Objective-C to figure this out. ¯\_(ツ)_/¯

reza-Abdollahi, 2019-04-28:
i have the same problem on iOS (iPhone6 & iPhone XR, ios12.x). but it seems that the problem is only with the portrait mode; by rotating the phone to landscape, faces would be detected correctly!

telmen, 2019-05-01:
I think this PR is responsible for this issue. After reverting this PR, face detection works on 2.6.0 https://github.com/react-native-community/react-native-camera/pull/2050

sibelius, 2019-05-01:
anybody can debug more this PR https://github.com/react-native-community/react-native-camera/pull/2050

we can revert this, but it be cool to have a solution that solves both cases

MateusAndrade, 2019-05-02:
> anybody can debug more this PR #2050
> 
> we can revert this, but it be cool to have a solution that solves both cases

I Will give a check on this

Genhain, 2019-05-16:
*Bump*
I am facing this issue currently, I used the mostly automatic pod installation and am finding that without changing any code the only time any results are returned using the front facing camera on an iPhone are when the device orientation is `UIDeviceOrientationLandscapeLeft`, no others work.

I also attempted @michaelcurtis changes with creating a `FIRVisionImage` directly from the sample buffer and had no luck.

I also attempted to simplify the `RNCameraUtils convertBufferToUIImage` code and with bare minimum it seemed to be rotated already? could the issue perhaps be the `VideoCaptureDevice` setup?

EskelCz, 2019-06-02:
For me it started working on ios (with 2.7.0) but doesn't work on android. (not sure about earlier versions)
Maybe there something missing in my installation? I would hope onFaceDetectionError would be called, but it's not, in my case.

n1ru4l, 2019-06-06:
@EskelCz Does it also work on 2.10.1?

closed 2019-06-06 by n1ru4l